Xinyuan-LilyGO/TTGO-T-Display #1: How to make the display work?

Nufflee, 2019-07-25:
I tried running the exact example code using the forked TFT_eSPI but the only thing I could get working is the backlight. 
I also used the correct User Setup but it just wouldn't work.
I could use some guidance with this.

tablatronix, 2019-07-25:
At first I thought it was tft_espi, but that will not have the offset defines in it yet.

I found this though

https://github.com/REVOtic/TFTDriver/tree/aa39c2bfaed3111db057056c94eef3a14bb92db6

Nufflee, 2019-07-25:
@tablatronix That actually does work perfectly! Thank you so much!

Nufflee, 2019-07-25:
Actually @LilyGO @lewisxhe you should fix the README.

cyraxx, 2019-07-27:
Looks like the repo linked in the README now includes the proper defines. It works fine for me (after enabling the correct User_Setup).

TotallyInformation, 2019-07-27:
@cyraxx I've just tried the version in the README and it does not work - see Issue #4

cyraxx, 2019-07-27:
Yup, both of the libraries referenced in README are actually forks.

I didn't check whether their fork or the REVOtic one is more current but they do both include the offsets needed for this board.

TotallyInformation, 2019-07-27:
This is what the readme says:

> Move or copy User_Setups/TTGO_T_Display.h to TFT_eSPI/User_Setups/

But that source does not exist in this repo nor in the fork. So the README needs to be at least updated.

I've tried using the original TFT_eSPI library and I can _nearly_ get it working correctly. However, the offsets are clearly not quite right somewhere because some of the original bitmap still shows to the right of the screen (about a 1/3 of the right-hand-side).

I'll look at it again tomorrow if I have time and if I can work it out, I'll raise an issue with Bodmer's repo.

cyraxx, 2019-07-27:
> > Move or copy User_Setups/TTGO_T_Display.h to TFT_eSPI/User_Setups/
> 
> But that source does not exist in this repo nor in the fork. So the README needs to be at least updated.

The source file actually is in this repo but the path in the README is wrong - it's in the root directory, not in `User_Setups`. Copying that file into the lewisxhe fork of TFT_eSPI (and editing `User_Setup_Select.h` to include it) made it work for me.

BTW I found this issue in Bodmer´s repo: https://github.com/Bodmer/TFT_eSPI/issues/379 I think that's the same screen.

lewisxhe, 2019-07-28:
Yes, the referenced TFT_eSPI is the branch that I modified after the fork. I should describe it clearly in the readme. Thank you for your reminder.

TotallyInformation, 2019-07-28:
Well, I've looked again but there are too many differences between the libraries for me to work it out. I'll raise an issue in Bodmer/TFT_eSPI though to see if someone with better understanding than me can add T-Display support to the original library.

I still can't work out how the offset works even though I've changed the entries in the TFT_Drivers folder to include the rotation (and other) changes.

tablatronix, 2019-07-30:
hmm I cannot get this to work , what pins did you use?

I have
```c++
#define TFT_MISO -1
#define TFT_MOSI 19
#define TFT_SCLK 18
#define TFT_CS   5  // Chip select control pin
#define TFT_DC   16  // Data Command control pin
#define TFT_RST  23  // Reset pin (could connect to RST pin)
#define TFT_BL   14  // LED back-light (only for ST7789 with backlight control pin)
```

tablatronix, 2019-07-30:
The backlight is IO4 NOT io14 as the graphic on the readme says!

Bodmer, 2019-08-02:
TFT_eSPI has been updated to include offsets for this display. A new [user setup file](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup135_ST7789.h) has been added to show the setup used for testing.

lewisxhe, 2019-08-06:
The library used has been synchronized to the main branch

SuperEugen, 2019-08-28:
I added the following two lines to the TTGO_T_Display.h file and can now use the examples from the original TFT_eSPI library from @Bodmer :

#define TFT_BL    4
#define TFT_BACKLIGHT_ON HIGH

@lewisxhe can you add these lines to the .h file?

Joennuh, 2019-08-28:
It is the best to handle that in your main sketch since the pin on which the led backlight is connected is capable of PWM. Defining an on or off state in header file could lead to issues to set the backlight intensity.
And besides that: the example sketch already turns the backlight on on line 135: https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TTGO-T-Display.ino#L135

Bodmer, 2019-08-28:
Defining the backlight pin and level in the setup file does not affect the ability to use PWM later in the sketch. The pin is only set during the init() function.

The point of defining the pin in the setup file is that all the existing [TFT_eSPI examples](https://github.com/Bodmer/TFT_eSPI/tree/master/examples) can then be run without modification.

lewisxhe, 2019-08-28:
The reason I define it in the sketch is to let everyone know the pin behind it.  I feel that this is not a big problem.

lewisxhe, 2019-08-28:
I can of course add it.  Small meaning. @SuperEugen

auculab, 2020-02-10:
Hello, can you pls request your support on how to turn off the backlight of TTGO T-Display. I want to turn off the display backlight after a timeout to be able to conserve battery if not being used and bring back at the press of button 2. I'm unable to find how to reverse this. 

>          digitalWrite(TFT_BL, TFT_BACKLIGHT_**ON**); 

Thank you

lewisxhe, 2020-02-11:
digitalWrite(TFT_BL, 0);  @auculab

kalyan540, 2020-03-23:
Pls help me the TTGO-TDisplay won't display anything after I upload a new code TTGO_T-Display.ino,and I follow instructions in readme.md,could any one rectify my problem.

lewisxhe, 2020-03-23:
@kalyan540 Please manually refresh the firmware in the warehouse to eliminate hardware problems.

kalyan540, 2020-03-23:
@lewisxhe Thanku for your reply
If my problem is not solved then I reply ones.

lewisxhe, 2020-03-23:
@kalyan540 If that doesn't work, please contact customer service and provide a video

kalyan540, 2020-03-23:
@lewisxhe it is working but not displaying anything ,it shows in serial monitor, and it is a new one before uploading it displays but after uploading only backlight is glowing ... What should I do..?

lewisxhe, 2020-03-24:
Sorry, it should be the screen is damaged, please contact customer service

kalyan540, 2020-03-24:
@lewisxhe Thanku for your reply
It is working with Adafruit_ST7789.h along with Adafruit_GFX.h library functions , I given a img in below, I think code has some errors, pls do any changes, and I got an error msg " TFT_BACKLIGHT_ON was not declared in this scope " Thanku.

lewisxhe, 2020-03-24:
@kalyan540 
Define a macro directly
`` `
#define TFT_BACKLIGHT_ON 1
`` `
Or directly change TFT_BACKLIGHT_ON to 1

kalyan540, 2020-03-25:
@lewisxhe Thanks for your reply
```
#define TFT_BACKLIGHT_ 1 //it is working (Backlight on)
```
But the TFT_eSPI.h library function is not working for my ttgo t-display ,and I followed the instructions in Readme.md 
Pls show me a way to work with this library .