Xinyuan-LilyGO/TTGO-T-Display #4: Install instructions do not work for Arduino IDE

TotallyInformation, 2019-07-27:
Using the Arduino IDE I followed the instructions in the README but there are some issues:

1. The Button2 library is not in the Arduino library manager, you have to download manually.
2. Then the compile fails with the following errors:

```
C:\src\arduino\TTGO-T-Display\TTGO-T-Display.ino: In function 'void button_init()':

TTGO-T-Display:50:30: error: no matching function for call to 'Button2::Button2()'

     pBtns = new Button2 [args];

                              ^

In file included from C:\src\arduino\TTGO-T-Display\TTGO-T-Display.ino:5:0:

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:55:5: note: candidate: Button2::Button2(byte, byte, unsigned int)

     Button2(byte attachTo, byte buttonMode = INPUT_PULLUP, unsigned int debounceTimeout = DEBOUNCE_MS);

     ^

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:55:5: note:   candidate expects 3 arguments, 0 provided

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:29:7: note: candidate: constexpr Button2::Button2(const Button2&)

 class Button2 {

       ^

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:29:7: note:   candidate expects 1 argument, 0 provided

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:29:7: note: candidate: constexpr Button2::Button2(Button2&&)

C:\src\arduino\libraries\LennartHennigs-Button2-4105e70\src/Button2.h:29:7: note:   candidate expects 1 argument, 0 provided

C:\src\arduino\TTGO-T-Display\TTGO-T-Display.ino: In lambda function:

TTGO-T-Display:62:26: error: 'TFT_DISPOFF' was not declared in this scope

         tft.writecommand(TFT_DISPOFF);

                          ^

TTGO-T-Display:63:26: error: 'TFT_SLPIN' was not declared in this scope

         tft.writecommand(TFT_SLPIN);

                          ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\myuser\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
no matching function for call to 'Button2::Button2()'

```

TotallyInformation, 2019-07-27:
The Button2 issue was fixed by using the lewishxe version - the link in that repo points to the original which doesn't work.

TotallyInformation, 2019-07-27:
Now, the remaining error is:

```
Build options changed, rebuilding all
C:\src\arduino\TTGO-T-Display\TTGO-T-Display.ino: In lambda function:

TTGO-T-Display:62:26: error: 'TFT_DISPOFF' was not declared in this scope

         tft.writecommand(TFT_DISPOFF);

                          ^

TTGO-T-Display:63:26: error: 'TFT_SLPIN' was not declared in this scope

         tft.writecommand(TFT_SLPIN);

                          ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\myuser\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
'TFT_DISPOFF' was not declared in this scope
```

TotallyInformation, 2019-07-27:
OK, so I'm guessing that the reason it isn't working is that you have referenced a FORK of the actual library instead of the library itself. That fork is **96** commits behind the original!

Why on earth would you do that? 

At least make it clear in the README that is what you have done.

lewisxhe, 2019-07-28:
Yes, the referenced TFT_eSPI is the branch that I modified after the fork. I should describe it clearly in the readme. Thank you for your reminder.

lilphil, 2019-07-29:
It would be a good idea to rebase from the original Bodmer repo. They have renamed the driver, so this patch works against the 1.4.11 release from that repo for the TTGO X Display. Although I'm not sure why it needs to be set as 240 width when it clearly isnt...

[1.4.11-TTGO-T-Display.txt](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/files/3443782/1.4.11-TTGO-T-Display.txt)

TotallyInformation, 2019-08-03:
Am I missing something? With all due respect, I can't see why this fork is needed at all? Why not PR any improvements back to the original rather than confusing people?

TotallyInformation, 2019-08-03:
@lilphil/@lewisxhe why do we need another fork? Much better to update the original (it has now been updated and should work just fine) than confuse people with another fork that gets out-of-date.

phillip-toone, 2019-08-05:
Okay, I am super confused here and have been working on this for hours.  Am I supposed to use the linked forks or the original? Neither work for me. The original just gives me errors and linked forks display nothing but the serial monitor shows me that it is working as expected. Also, why are there instructions about moving and modifying files. Shouldn't those details be taken care of in the fork? I don't know how I did it but I once had a display that read "Voltage :4.70" with a small rectangle next to it that seemed to be a shrunk down version of what it used to display on the entire screen what I first purchased it. Also, the pretty graphical TTGO display does not show. Help please.

FixxCZ, 2019-08-05:
Check this video on how to install it https://youtu.be/qj9dN-Ginxc

phillip-toone, 2019-08-05:
Works! Thanks.

lewisxhe, 2019-08-06:
The library used has been synchronized to the main branch

spradlinb, 2019-09-13:
I'm having problems using this repository with the TTGO T-Display module. I've followed the instructions in the ReadMe file, installed the latest versions of the master branch for both TFT_eSPI and Button2 libraries. I've downloaded this repo's files. I've moved the TTGO_T_Display.h file into the User_Setups folder, and referenced it in the User_Setup_Select.h file. I've made sure the bmp.h file is in the correct folder to build and push the files to the device. However, when the code runs it is only updating about 1/3 of the screen, and the text horizontal, not vertical. Something seems to be wrong in the current files and I'm not sure what I need to change to get it working again. Help would be greatly appreciated!

spradlinb, 2019-09-13:
For others that might still be having problems with the screen: If it looks like only 1/3 of the screen is being used, be sure to comment out the default "#include <User_Setup.h>" line of text in the User_Setup_Select.h file that you add the "#include <User_Setups/TTGO_T_Display.h>" line to. Keeping both lines will cause the screen settings to be incorrect. This was not shown in the youtube video linked by FixxCZ, but solved the problem for me.

Mystereon, 2019-09-26:
https://github.com/Mystereon/TTGO-T-DISPLAY-LIFE-/tree/master

flyingdan, 2019-10-23:
Just worked through the instructions making sure to follow the video and @spradlinb 's suggestion and the image displayed on the screen is half the TTGO image and half the multicolored screen. Here's an image of how it looks 
[image]

Any thoughts?

ustolt, 2019-10-24:
In 
_https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h_ must
ILI9341_DRIVER be commented out.

```
// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
```

tochinet, 2020-01-03:
iCompiles and upload for me (I trashed the original WiFi library), but nothing on screen, the sketch doesn't even start, and I get this in the Serial monitor : 

**Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
assertion "false && "item should have been present in cache"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t)
abort() was called at PC 0x400d8f3b on core 0

Backtrace: 0x400884bc:0x3ffcd750 0x400885bb:0x3ffcd770 0x400d8f3b:0x3ffcd790 0x400dd78c:0x3ffcd7c0 0x400dde42:0x3ffcd7e0 0x400de171:0x3ffcd830 0x400dd23c:0x3ffcd890 0x400dcd96:0x3ffcd8e0 0x400dce2f:0x3ffcd900 0x400dce7a:0x3ffcd920 0x400d4ca8:0x3ffcd940 0x4010ff47:0x3ffcd960 0x400da60a:0x3ffcd990

Rebooting...**
and again and again

Actually, it looks similar to the heap problems I 've found on other forums that make the ESP crash in special WiFi cases, but in my case I don't even see  the first print statement in setup().

Any idea what could be wrong ? I updated my ESP32 files to 1.0.4 but it doesn't change anything

Achref-omrani, 2024-02-06:
> Using the Arduino IDE I followed the instructions in the README but there are some issues:
>
> 1. The Button2 library is not in the Arduino library manager, you have to download manually.
> 2. Then the [...]

did you have any errors such this one 

Compilation error: bmp.h: No such file or directory

if it is a library i can find it anywhere