Xinyuan-LilyGO/TTGO-T-Display #6: PWR EN functionality

paccerdk, 2019-08-26:
I've been trying to figure out the functionality the PWR EN, but your schematics is difficult to figure out in this regard, and its not documented anywhere.

As far as i could figure out, the PWR EN "pin" is supposed to shut off the 3.3V regulator when powered by battery (and usb is disconected), but it does not seem to work (i've tried toggling it)

Joennuh, 2019-08-26:
Just a guess: maybe it is read only? So you can read the value and let your sketch find out in this way whether it runs from battery or USB power?

paccerdk, 2019-08-27:
@Joennuh its an output, there's a diode on the pin, restricting current flow to one direction.

gnunomad, 2019-09-22:
I have the same problem. I've been analyzing the scheme but I can't figure out how that pin works. I think the scheme contains errors in the polarity of the mosfet.

lewisxhe, 2019-10-08:
I get feedback: PWR_EN can continue to detect the power of the lithium battery when there is no USB power supply. V3V is to prevent leakage current. @paccerdk

paccerdk, 2019-11-17:
@lewisxhe
This isn't making much sense to me.
Detect in which way? As mentioned earlier, it appears to be an output, there is a [diode](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf) restricting the current flow to one direction. (out)
Could you perhaps share the feedback that you got?
Honestly, it seems to me like there is a mistake on the schematic, and the PWR_EN was supposed to be able to let the device turn itself off, when battery powered. that also explains the naming.

lewisxhe, 2019-11-18:
@paccerdk 
In order to use VBA to turn on the power, because, when downloading 32, the single chip PWR_EN can not work, so V3V has no power, so 2104 can not be downloaded, so open V3V power through VBUS

The above is the original words, I am responsible for the transmission.

ricolk, 2020-03-17:
I cannot understand BAT offsheet. It is just connected to  Q4 and to the partition for battery conversion... Is it connected also to VBAT?

ticksp, 2020-04-01:
[ESP32-TFT(6-26).pdf](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/files/4416932/ESP32-TFT.6-26.pdf)
For understand PWR_EN functionality you can have a look on comments, which I made in attached schematic.
So you can set PWR_EN to "1" and read voltage in BAT_ADC, if this voltage more than 4.3 V device have powered from USB. If less then 4.3 V - device have power from battery. To save battery you can set PWR_EN to "0" and in this case USB converter will be power off and do not use your battery. When you need to measure battery voltage first set PWR_EN to "1", measure voltage and then set PWR_EN back to "0" for save battery. I hope my descrition will help somebody to better use this device! :)

pvonmoradi, 2020-10-24:
@ticksp Thanks for the clarification. So the overloading of GPIO14 (as it is JTAG's TMS pin) does not hurt JTAG functionality right? 
The way I see it, while debugging via JTAG, some transistors would get stressed (because of PWR EN) and the "BAT_ADC" voltage is not to be trusted, but nevertheless, this wouldn't hurt debugging. 
#17 #26

tadam777, 2021-01-31:
It's possible to solder a wire on the bottom of the T-Display module to access the GPIO14. I didn't test it but here's where to solder :

[image]

posplayr, 2021-06-24:
Can anyone confirm that you can access IO14 at D6 in order to support JTAG and have the JTAG work? 

Also it seems as if from ticksp's comment above:
**_When you need to measure battery voltage first set PWR_EN to "1", measure voltage and then set PWR_EN back to "0" for save battery._** 

if not using a battery then D6 be removed and this makes it easier to solder a wire to the anode pad of D6?

kiromaisus, 2021-08-15:
Yes, it's possible to solder a pin on one end of the diode that's connected to IO14 and use JTAG. I did it with ESP-prog and it worked. Not needed to remove the diode, but I'm not using it with battery, just to debug.

posplayr, 2021-08-18:
> Yes, it's possible to solder a pin on one end of the diode that's connected to IO14 and use JTAG. I did it with ESP-prog and it worked. Not needed to remove the diode, but I'm not using it with battery, just to debug.

Thank you for that confirmation.