Xinyuan-LilyGO/TTGO-T-Display #9: Auto power-on when powered trough battery or 3.3v/5v pins

oniongarlic, 2019-10-03:
Is it possible to get the board to power-on automatically when powered trough the battery or 3.3v/5v pins ? At least my board requires a press of the reset button for it to start up.

When powered trough USB-C it turns on automatically, but unfortunately I can't use that in the final installation.

Joennuh, 2019-10-03:
It is actually too bad that the device hasn't got an reset-pin. When it does have an reset-pin you could attach an external reset-button to it.

lewisxhe, 2019-10-08:
This problem has been feedback, you can remove the marked position capacitance in the figure can be solved @oniongarlic 
[image]

oniongarlic, 2019-10-10:
Thank you, I will try that.

smartynov, 2019-12-01:
@oniongarlic did it help you?

Jeopardye, 2019-12-08:
Can confirm, this works. (Which is good, because there is no way I could solder this back in. The brute force from the screwdriver probably didn't help much either...)

cc @smartynov

austrisv, 2020-01-12:
Did help for me as well ... for one or two reboots. Then suddenly it was not connecting to wifi anymore. Checked the console and upon reset can see "waiting for download" and
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))

If I do upload from Arduino - it succeeds and then this. As if I would be holding gpio0 to ground on restart (where I even can't spot the gpio0 on the board).

Dear @lewisxhe, could it be related w removal of capacitor? Any other ideas what could have gone wrong? Thank You!

lewisxhe, 2020-01-13:
@austrisv Because it is GPIO0 to control the download and start, there is only the reason for this capacitor. This is the conclusion that I use the battery to connect the board and the UART output.

kostashellas, 2020-01-23:
what's the solution to this issue?
for my project I need power on when 5.3v connected to +5v pin (works only if RESET button triggered) and WiFi connection right after the power on.
C3 cap 100nF can be reduced to have both functions available?

[image]

john4242, 2020-02-11:
Same issue. After looking at the schematic, I'm not sure the VBUS (USB) doesn't have this power issue either, maybe the USB-TTL chip is sending an RTS signal on power or computer connection, and that is doing the reset?
CHIP_PU seems to be the reset pin on the schematic?
[image]
[image]

hpsaturn, 2020-05-09:
Same issue for me, after some hours, with the USB-C connected and the application running, it turn off. The only mode for turn-on it is with the reset button. It is a very bad behaivor for aplications 24/7 :'(

spiff72, 2020-05-28:
Has there been any additional info on this topic.  This is still a struggle for me.

I have several of these boards and I want to power them from the 5v pin (no battery needed).  But if they are disconnected from power, pressing the reset button to restart them is a pain.

I even see different behavior between boards (with different code loaded).  One seems to start up fine when 5v is connected to the pins, and the other goes into a state where it is drawing about 20ma of current and won't boot until the reset button is connected.  

Can the code itself somehow prevent this behavior?  Or the bootloader?  The one that absolutely refuses to start without the reset button push has been loaded with a bootloader i donwloaded from a support thread from the WLED developer, so this makes me wonder if that might be contributing to the issue in my case.

spiff72, 2020-05-28:
UPDATE:  I just flashed the bootloader to the board that seems to start up correctly with the 5v pins, and after doing so, it still operates normally (no push of the button needed).

tehcaster, 2020-06-01:
> UPDATE: I just flashed the bootloader to the board that seems to start up correctly with the 5v pins, and after doing so, it still operates normally (no push of the button needed).

So what you're saying is, if the board boots on 5V pin without pressing reset, flashing the bootloader will not break that. But also if the board does *not* boot without pressing reset, flashing the bootloader also will not change that. Correct?

So the question is, what makes the difference between boards that boot and boards that don't? Of those few I have, none boots without reset. It's really a setback on otherwise great board, I wish the reset was included in the pin header. Did anyone e.g. try to desolder the built-in button and wire it away from the board?

spiff72, 2020-06-01:
> So what you're saying is, if the board boots on 5V pin without pressing reset, flashing the bootloader will not break that. But also if the board does _not_ boot without pressing reset, flashing the [...]

I have an update to my previous work...
I ended up building soldering up a board (in a printed enclosure) with this TTGO T-Display over the weekend.  I am running the WLED project on it (https://github.com/Aircoookie/WLED/) - which I submitted a pull request for over the weekend as well to roll in a usermod that allows for the use of this TTGO T-Display with that WLED project and Platformio.
My arrangement is as follows:
- 12v power supply plugs into a barrel connector in the enclosure
- The 12v power supply passes through to the addressable LED strip I am using (WS2815), and I also have a small buck converter in the enclosure that drops the voltage down to 5v to power the TTGO board and a level shifter (for bumping the data signals of the TTGO up to 5v to feed the LED strip).

I just unplugged it 4 times in a row, and plugged it back in, and it powered up successfully (fully booted) without any need for a push of the reset button.

I can't recall for certain which of the boards I used for this build, but I am fairly confident that it was the one that was loaded with the clock program I referenced earlier in this thread.

I have no explanation as to WHY this is now booting without issues.  I think I did notice just one instance where it didn't boot up after initial connection to power (right after i built it), but unplugging the 12v power and plugging it back in allowed it to fire up just fine.

The only difference between this situation and the one I was testing before is that I am powering it with the 12v --> 5v buck regulator instead of my Agilent E3646A bench supply (at 5v).

This may not help others very much - but I just wanted to share my results...

sej7278, 2020-09-20:
So what's the solution to this? Seems removing the capacitor stops it from booting or something? Lots of gibberish in this thread!

sej7278, 2021-04-08:
@lamello have you tried the lipo connector?

spiff72, 2021-04-08:
Just to chime in here again, I have built several more of these WLED controllers using this board and have not had issue using the approach I described above.  I did not modify the boards in any way and they "just work" in my case.  Not sure why others don't...

lamello, 2021-04-08:
No i didn't try the lipo connector (because i will not use it in my project). As i said before i did powered directly the board with 3.3 Volt and the board won't start either.
When i look at the schema, i have a susspicion of the CP2104. The difference between powered by USB and direct (5v/3.3V) is that the CP2104 has power with the USB and not with the direct voltage. The CP2014 controls the DTR and RTS and these lines has something to do with the CHIP_PU signal. I have programmed now for several months for the project and the program  begins to get really nice, so this hardware issue is a bummer. I use almost no pins (just i2c and an pin for interrupt. I really want to use this board because is is compact and has a beautiful little display. I will look further for the issue, but i have now no problem to connect the CHIP-PU signal to an unused IO-pin and to externaly reset it (maybe with an attiny). Will see.

sej7278, 2021-04-08:
I wouldn't use a TTGO display in a final product it's just too unreliable and no two boards seem to work the same.

I'm currently prototyping on the Display  then moving to a Mini with a separate tft for the finished product

lamello, 2021-04-08:
@sej7278: What do you mean by "a Mini". I have worked with a mini, but that was a DEC PDP11. ;-)

sej7278, 2021-04-08:
TTGO T7 Mini https://m.aliexpress.com/i/32977375539.html

lamello, 2021-04-15:
Thanks for sharing.

I haven't solved it, but have now a workaround. I have made a wire from the reset switch to Pin26 (which is not used in the project).
I have programmed an attiny85 which generate a 150ms negative pulse at his startup. The attyny is powered by the 3.3Volt of the TTGODisplay. Connect the pulsepin of the attiny to pin26 of the TTGODisplay with an 120 Ohm resistor and voila, the TTGO boots when it is powered with 5Volts.
I know, it's overkill, it won't win the beauty contest. But it boots now.
I'm done with it.

Greetings lamello

trycoon, 2022-09-18:
I can confirm that unsoldering the capacitor works for three of my TTGO-T1 boards, that normally require the reset button to be pressed to power on. This is when powering the board from the 5V pins.
A fourth board works without this hack, so there may be different revisions of the board.

CLGamesStudio, 2023-03-09:
Help me please! I flashed the board several times, but after loading the Tetris game on it, it stopped being displayed on the device panel, when the PC board is connected, it sees it, but when I try to load a new program, the com port disappears from the list. Thank you in advance