Xinyuan-LilyGO/TTGO-T-Display #10: GPIOs dont work as generic

arcbtc, 2019-10-05:
Using 38-27 for a 4x4 keypad, but they dont seem to work as generic ESP32 GPIOS
Using "Keypad" library
Same code works fine with DEVKIT V1 and NODEMCU 32S
```
const byte rows = 4; //four rows
const byte cols = 3; //three columns
char keys[rows][cols] = {
  {'1','2','3'},
  {'4','5','6'},
  {'7','8','9'},
  {'*','0','#'}
};
byte rowPins[rows] = {27, 26, 35, 33}; //connect to the row pinouts of the keypad
byte colPins[cols] = {32, 39, 38}; //connect to the column pinouts of the keypad
Keypad keypad = Keypad( makeKeymap(keys), rowPins, colPins, rows, cols );
```

arcbtc, 2019-10-05:
Pin 38, 39 seem to be the issue

lewisxhe, 2019-10-08:
@arcbtc 38,39 can only be configured as an input. If you confirm that it is pressed by detecting low level, please use a pull-up resistor to pull the PIN high externally.

billmetz, 2020-01-30:
From the IO_MUX table of the ESP32 Datasheet (V3.3), GPIO34 to GPIO39 are input only.  Perhaps it would be helpful to update TTGO-pinmap.jpg as :
[image]

tadam777, 2021-01-31:
You're right pins GPIO34 to GPIO39 are input only.
[image]

[Link to the ESP32 datasheet](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjiiJDfpMbuAhUQwAIHHZQXCDwQFjAAegQIAxAC&url=https%3A%2F%2Fwww.espressif.com%2Fsites%2Fdefault%2Ffiles%2Fdocumentation%2Fesp32_datasheet_en.pdf&usg=AOvVaw0aPqs26KTwgQAQ33yEvdPB)

xpix, 2021-04-12:
Well, what to say about it. A PCB design for the garbage can ... thanks for the tip to all.

yngndrw, 2021-06-17:
I've just fallen into this trap as well, like you I now have a load of useless PCBs.

It's clearly labeled as GPIO on the pinout so I'm very annoyed by it.

stevempotter, 2022-02-25:
You can unsolder a couple of tiny caps on the underside to get those GPIOs back. (C49, C32)