Xinyuan-LilyGO/TTGO-T-Display #11: How does the touch function work

EthanGe77, 2019-10-25:
I saw touch2-5 touch7-9 on the layout image. not quite sure how to use them

lewisxhe, 2020-01-08:
There are examples in Arduino esp. Just write the code and touch the IO port after running. You will see the response. 
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Touch

bbogdanmircea, 2020-01-24:
In the original code, there are 2 Button 2 instances attached to pins 35 and 0, but these pins are not on the board. How or where should I connect to these pins?

m-stefanski, 2021-11-21:
For those who will stumble upon this question. 

GPIO 0 and 35 are physical buttons that have to be clicked, TOUCH GPIOs are supposed to be, well, touched. Change of voltage upon touching gpio with a finger on the pin gets detected.