Xinyuan-LilyGO/TTGO-T-Display #12: Can I use any pins as rx/tx for softwareserial?

ykolerov, 2019-11-06:
Hi, I'm struggling to connect Sim800l to this board. Using arduino environment and tinyGSM via softwareserial (as no hardwareserial is pinned-out and available at this particular board? or?..) Tried different pin pairs as rx/tx with no success.

lewisxhe, 2019-11-18:
You only need to specify RX and TX at initialization time, as follows:
```
Serial1.beign(115200, SERIAL_8N1, 25, 26);
```

ykolerov, 2019-11-23:
Thanks.
I realized I have to connect modem RST to VCC. Now everything works.

closed 2019-11-23 by ykolerov

Eddiiie, 2021-03-17:
Hello,
Just want to confirm.   I have a project where I need 2 UART ports and also the USB Serial port, total of 3.

I put this test code together but have not uploaded it yet ..    There are no errors on compile check!   

Should this really work?   If so, that is great!  It just seems too easy..    Serial, Serial1, Serial2

I also have to level shift this to use one of those TTL serial ports with the DB9 connector, right?  They are 5V.

```
void setup() {
 Serial.begin(115200);
   while (!Serial) {}

 Serial1.begin(115200, SERIAL_8N1, 12, 13);
   while (!Serial1) {}

 Serial2.begin(115200, SERIAL_8N1, 2, 15);
   while (!Serial2) {}
}

void loop() {
  // put your main code here, to run repeatedly:
}

```