Xinyuan-LilyGO/TTGO-T-Display #14: SPI sensor

DavidAntonin, 2019-11-23:
It is possible connect external SPI sensor - C1101 433MHz transceiver?
On generic ESP32 I use pins: 
18	(SCK)
19	(MISO)
23	(MOSI)
5	(SS)
Thank you

lewisxhe, 2020-01-08:
Since 14pin of HSPI is already used, the following combinations can be used
CS 15
CLK 25 or 26,27
miso 12
mosi 13
 
21,22 is also possible, but generally used as SDA SCL for I2C

ketan, 2020-02-11:
This pin mapping definitely did not work.

```cpp
#define PIN_MISO  12
#define PIN_MOSI  13
#define PIN_SCK  26
#define PIN_CS  15
```

I'm getting the following error, it appears that the esp32 is unable to read its own memory, unplugging the microsd module seems to atleast print some debug logs via serial.

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

Has anyone managed to get an sd card to work with this TTGo board, could you share your pin mappings?

ketan, 2020-02-11:
Found the answer, the following pin mappings worked for me:

```
#define PIN_MISO  27  // GPIO27
#define PIN_MOSI  26  // GPIO26
#define PIN_SCK   25  // GIPO25
#define PIN_CS    33  // GPIO33

...
SD.begin(PIN_CS, PIN_MOSI, PIN_MISO, PIN_SCK)
```

romatou18, 2020-03-03:
> Found the answer, the following pin mappings worked for me:
> 
> ```
> #define PIN_MISO  27  // GPIO27
> #define PIN_MOSI  26  // GPIO26
> #define PIN_SCK   25  // GIPO25
> #define PIN_CS    33  // GPIO33
> 
> ...
> SD.begin(PIN_CS, PIN_MOSI, PIN_MISO, PIN_SCK)
> ```

I found out the same thing: trying to use the remaining available HSPI pins i.e. 15,13, 12 does NOT work at all. Tried it very hard, the only thing that works is using other generic GPIOS with the VSPI, although it's not as good as HSPI since the max clock speed is only 26MHZ if im correct vs. 40MHZ for HSPI. But using an SD card reader on SPI seems to work great, decent writing speed, way enough for logging text. 

If it was coming to writing a video stream real time from a camera that would likely be an issue. Anyway other modules like the TTGO Camera module are made for that. 

Now the big problem that remains is the fact that PIN 14 is the MS JTAG pin and therefore this is impossible to do JTAG debugging on TTGO-T Display. I found that out too late. Not an issue for production once the program works ok. But for development work this is a real issue. I am now thinking of developing my project on a different board.

RideOn123, 2020-05-09:
> Found the answer, the following pin mappings worked for me:
> 
> ```
> #define PIN_MISO  27  // GPIO27
> #define PIN_MOSI  26  // GPIO26
> #define PIN_SCK   25  // GIPO25
> #define PIN_CS    33  // GPIO33
> 
> ...
> SD.begin(PIN_CS, PIN_MOSI, PIN_MISO, PIN_SCK)
> ```

Which SD Libary did you use? The "standard" Lib expects different arguments for SD.begin().

ketan, 2020-05-09:
I'm using Arduino api on esp32:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.h

minkione, 2020-05-28:
I am trying to get working a CC1101 with my T-Display.
Someone suggested to device the following pins as SPI for teh connection between the T-Display and the NRF24...
MISO: 27
MOSI: 26
SCK: 25
CSN: 33
GDO0: 2
GDO2: 37
But it still doesn't work.

Anyone has some suggestions?

lewisxhe, 2020-06-28:
@minkione 

As shown in the code below, did you specify SPI to initialize it?
```
#include "FS.h"
#include "SD.h"
#include <SPI.h>

SPIClass SPI1(HSPI);

 

#define MY_CS     15
#define MY_SCLK   25
#define MY_MISO 26
#define MY_MOSI 27
void setup()
{
 

    SPI1.begin(MY_SCLK, MY_MISO, MY_MOSI, MY_CS);

  //Assuming use of SPI SD card
    if (!SD.begin(MY_CS, SPI1)) {

        Serial.println("Card Mount Failed");

    } else {
        Serial.println("DEU CERTO INICIAR SD CARD");
    }
}
```

minkione, 2020-06-28:
Yes, I defined them in my sketch:
```
//ELECHOUSE_cc1101.setSpiPin (SCK, MISO, MOSI, CSN);
ELECHOUSE_cc1101.setSpiPin(25, 26, 27, 15);  //Default LSatan Configuration For T-Display
```

In case you wanna try with a CC1101, I used this sketch 
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/Rc-Switch%20examples%20cc1101/ReceiveDemo_Advanced_cc1101/ReceiveDemo_Advanced_cc1101.ino

from the https://github.com/LSatan/SmartRC-CC1101-Driver-Lib

minkione, 2020-06-28:
For instance, I tried with different boards, ESP32 Wroom, ESP8266, UNO R3, MEGA2560.
The only board so far that doesn't work and cannot figure out why the SPI pins do not work... is the T-Display one...

minkione, 2020-06-28:
I have even added this:
  SPIClass SPI(HSPI);

here:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/ELECHOUSE_CC1101_SRC_DRV.cpp#L69
```

/****************************************************************
*FUNCTION NAME:SpiStart
*FUNCTION     :spi communication start
*INPUT        :none
*OUTPUT       :none
****************************************************************/
void ELECHOUSE_CC1101::SpiStart(void)
{
  // initialize the SPI pins
  pinMode(SCK_PIN, OUTPUT);
  pinMode(MOSI_PIN, OUTPUT);
  pinMode(MISO_PIN, INPUT);
  pinMode(SS_PIN, OUTPUT);
  
  SPIClass SPI(HSPI);
  
  // enable SPI
  #ifdef ESP32
  SPI.begin(SCK_PIN, MISO_PIN, MOSI_PIN, SS_PIN);
  #else
  SPI.begin();
  #endif
}
```

minkione, 2020-06-28:
And to be sure... I also tried by replacing all existing occurrences of "SPI." with "SPI1."

https://gist.github.com/minkione/795307620dbb93723678fcec3ff19368

No luck in getting CC1101 working...

lewisxhe, 2020-06-29:
I have tried using the SD card and everything is working, the code and README have been updated

closed 2020-06-29 by lewisxhe

pati5000, 2020-07-08:
@minkione you swapped the MISO and MOSI pins by mistake.

> Yes, I defined them in my sketch:
>
> ```
> //ELECHOUSE_cc1101.setSpiPin (SCK, MISO, MOSI, CSN);
> ELECHOUSE_cc1101.setSpiPin(25, 26, 27, 15);  //Default LSatan Configuration For T-Display
> ```
>
> In case you [...]

See the original post from @ketan 

> Found the answer, the following pin mappings worked for me:
> 
> ```
> #define PIN_MISO  27  // GPIO27
> #define PIN_MOSI  26  // GPIO26
> #define PIN_SCK   25  // GIPO25
> #define PIN_CS    33  // GPIO33
> 
> ...
> SD.begin(PIN_CS, PIN_MOSI, PIN_MISO, PIN_SCK)
> ```

Hope this helps. All the best

flyandi, 2020-07-29:
So for anyone still having issues. Both HSPI and VSPI are working BUT you need to ensure you connecting  it right. I had issues with the actual pin-out of using an TF Card Holder and just "blindly" followed the pictures. Turns that both are wrong.

Here is the correct pin out on both the Micro SD and the adapter:

[image]

Biggest feedback is there that PIN 1 is NOT CS.. I repeat PIN 1 is NOT CS .. PIN 1 is not used for SPI and PIN 2 is CS .. after that it worked without flaws both on HSPI (40mhz) and VSPI (26mhz) ...

tadam777, 2021-12-12:
> 

Just confirm it's possible to JTAG debug by soldering the required IO14/TMS pin under the board : https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/6#issuecomment-770394385