Xinyuan-LilyGO/TTGO-T-Display #17: is there a way to connect JTAG debugger to the processor on this board ?

anonymouze, 2020-01-11:
I can't see any JTAG pins in the schematics. Need to set up a debugger.

lewisxhe, 2020-01-13:
Unfortunately, JTAG cannot be used for debugging

bastianhjaeger, 2020-05-28:
Is there any other way of debugging? E.g. via the USB?

pvonmoradi, 2020-10-24:
@lewisxhe Out of 4 required JTAG pins, only GPIO14 (MTMS/TMS) is not available. What is this "PWR EN" that is connected to it? Following the schematics, it seems it is attached to D6 diode.

tadam777, 2021-01-31:
To test but the pin GPIO14 can be accessed under the board : https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/6#issuecomment-770394385

robertlipe, 2021-03-14:
If you need JTAG in this form factor, the Longan Nano does have JTAG brought opposite the USB connector, making it a real pain on breadboard since 

I wish I'd noticed lewisxhe's comment before, as I ordered this to step up (over?) from Longan Nano just in the hopes of having a better video display controller. That sentence really puts the hurt on using this in development. 

I was just told by B*d that this device was discontinued. I can't tell if that's from them or by the maker, but they DO seem to be getting rarer.

Off to Plan D for an inexpensive, self-contained RISC-V development/debugging environment with a screen.

robertlipe, 2021-03-14:
@lewisxhe , now that I look at it, are you sure? I'm looking at the Longan Nano and my (dysfunctional) T-Display and all the JTAG pins are in the same place.  In the GD32V Manual (500+pages), section 7.4.3 says the packages with low pin count (e.g. the one used here) can be reconfigured to remap the JTAG from (non-existent) pins.  The mappings are exactly the silk screens on this board and in the same location as they are on the Nano. 7.3.1 and 10.2.1  says the startup values of the GPIO lines out of reset should work for JTAG.

7.5.9 shows that SWJ_CFG[2:0] defaults to remapping being on and "The default state after reset is SWJ
ON.This allows JTAG mode to be enabled by sending a specific sequence on the JTMS/JTCK pin."

I wonder if this really IS the same as the NANO and it's just a silkscreen labeling issue that makes it look like JTAG isn't available. 

TMS is one of those that gets remapped.

Can anyone with a working board try the pinouts from https://wiki.segger.com/SiPeed_Longan_Nano or offer a definitive reason why this doesn't work?