Xinyuan-LilyGO/TTGO-T-Display #20: Unable to read the battery voltage when USB is disconnected

yellowbeak, 2020-01-23:
When power is connected via USB, I can read the ADC value on pin 34, which is the BAT_ADC signal on the schematic and is half the supply voltage.  

However, if I connect a LiPo battery, and disconnect the USB, this pin always returns a value of 0.

I have tried setting pin 14, which is PWR_EN as an output, and setting both HIGH, and LOW, but neither of these change the value I'm reading from pin 34 - it always reads 0 when running from battery.

How can I read the battery voltage when running from the battery without the USB connected?

bbogdanmircea, 2020-01-24:
I have just tried this board yesterday with the code from master and it shows the battery voltage on the display even when USB is not connected.
What code are you running on the board?

yellowbeak, 2020-01-24:
Sorry - you are correct.  I went back to the original example and it works as expected.  I was using a modified version of this, that incorrectly reconfigured one of the pins by trying to read it as a touch input, which gave the problem that I was seeing.  Once I removed this, it all works as expected.
Please delete this issue thread.

closed 2020-01-24 by yellowbeak