Xinyuan-LilyGO/TTGO-T-Display #21: Function of LED

jmarchewka, 2020-02-07:
In looking at the schematic, the LED (it says RED but mine is blue?) seems to be connected to the CHRG line on the TP4054 LiPo charger IC. If I am reading the datasheet of the TP4054 correctly, the CHRG will go high once the charge cycle is complete, which should cause the LED to turn off. I have left my 1600mah LiPO battery connected for 36 hours and the LED never went off. Is my assumption wrong?

mhannebert, 2020-04-13:
I'm not sure exactly where the blue LED is connected. The thing is mine turns on when it is charging - and off when fully charged.

radiohe4d, 2020-11-11:
Mine is also a blue LED. I'm not using a battery, just running off the USB-C power.

Its on all the time and never goes off.

Is it possible to control this blue LED from the code?

nanosek, 2020-11-28:
I would also be interested how to turn off this blue LED

Batlapin, 2021-03-13:
"Is it possible to control this blue LED from the code?"
Nope.