Xinyuan-LilyGO/TTGO-T-Display #26: PWR_EN on IO14

mhannebert, 2020-04-13:
I'm not sure what the "PWR_EN" is for...

It is connected to IO14
In the demo code, there is  a "#define ADC_EN 14" but this ADC_EN is not used further...

It is related to the ADC ? I don't think so. To measure the voltage of battery, this is IO34.
It looks like it is triggering one MOSFET which is linked to the battery?

mhannebert, 2020-04-15:
The demo code was updated in this respect just a few days after my comment above.

It looks like the PWR_EN set to HIGH enables to measure the batterie itself. Otherwise, it measures the voltage provided by USB? I'm not sure.

Not tested yet.

pvonmoradi, 2020-10-24:
Same question here. Since GPIO14 is JTMS, this essentially causes the board to lose support for JTAG debugging.  
[image]
[image]

tadam777, 2021-01-31:
It's not too complex to access this pin and/or remove the diode : https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/6#issuecomment-770394385

bbogdanmircea, 2021-02-16:
Can anybody explain according to the schematic what happens when this PWR_EN is high or low?

jej, 2021-02-27:
Here: #6