Xinyuan-LilyGO/TTGO-T-Display #27: How to power the board?

mhannebert, 2020-04-13:
I want to power the board without the USB.

Should I provide 5V to the "5V" pin?
Should I provide 3.3V to one of the "3V3" pins?
Would any of the above work?

I guess that if I provide 3.3V, the battery won't charge. But will it charge if I provide 5V?
It looks to me that only USB, through the TP4054, would enable the battery to charge...

M

JoeBarouneD, 2020-04-13:
Like any ESP32, you "normally" can power it by 5V or 3.3V.
I don't know if it can "step up" from 3.3v to Vcharg, you can try and report the result.
But i think it will work without problems on 5V.

flyandi, 2020-04-15:
I run it off a 1s lipo without issue.

spiff72, 2020-05-27:
Rather than attempting to create a new issue - I am looking for some help here.

In order to power this TTGO T-Display board with a voltage source (not through the USB-C connector), how would one go about this?  

Are you connecting 5v to the 5V pin?  Or 3.3V to the 3.3V pin?  The 3.3v pin appears to be the output of the 3.3V regulator, so I am not sure that is the right approach.

And from my (unprofessional) look at the schematic, it seems like a 5V source would ideally be connected to the VBUS line, but that isn't broken out to a pin.

Am I missing something here? 

I was hoping to use these for a project where I mount it in an enclosure, with a 12V input from a wall wart, and a small buck regulator to drop the voltage to 5v to power this and a level shifter for controlling some 12v addressable LEDs.  But not being able to power from that 12V source makes this board a bummer for me.  Even if it could be externally powered with a 3.3v source, I could put a second buck regulator in there for the 3.3v source and make it work.

But I have also seen some mentions that the board doesn't start "automatically" when power is applied through a method outside of the USB-C port.

JoeBarouneD, 2020-05-28:
Powering it by the JST connector is working (you can put a 18650 bat to test).
I could power it by USB-C with some voltage range.

Powering it by the 5V should work.
Just try it.

sej7278, 2020-09-20:
I can't get the battery connector on the underside to turn on the display, it seems to run a IR-LED i have hanging off the 5v pin (and I assume the ESP32 itself) but I can only get the display to turn on when running from USB-C. Does the display need 5v?

ribbles, 2020-10-31:
If I power via 5v rail, 5.0v is sent to the LiPo battery, so it looks like it is passing battery management. If I power via USB-C, 4.2v is sent to the LiPo battery. So I am unable to use battery management unless using the USB-C port, since there is no VBUS connector. I guess I could remove the USB connector and solder on to that, but that just makes it difficult to improve the software later on.

@sej7278 - I am able to run the controller & screen using just a LiPo battery plugged into the JST port, and also via the 3.3v rail.

I am using this schematic for reference: [https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO-T-Display/master/schematic/ESP32-TFT(6-26).pdf](https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO-T-Display/master/schematic/ESP32-TFT(6-26).pdf)

sej7278, 2020-10-31:
@ribbles when you run the controller and screen using a lipo, do you need to press the reset button to make the screen come on?

ribbles, 2020-11-01:
No. But my board was shipped without a capacitor next to the USB-C port, which I read is what needs to be removed here: [https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/9](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/9)

madameczek, 2021-03-10:
> If I power via 5v rail, 5.0v is sent to the LiPo battery, so it looks like it is passing battery management. If I power via USB-C, 4.2v is sent to the LiPo battery. 
...
> I am using this schematic for reference: https://raw.githubusercontent.com/Xinyuan-LilyGO/TTGO-T-Display/master/schematic/ESP32-TFT(6-26).pdf

Are u sure that 5V from rail is passed to a batterry connector or U6 batt management chip? I didn't test, but can't see this from schematic.

ismailagir, 2021-04-18:
Do 3.3V output also works when a lipo is connected?

ribbles, 2021-04-19:
Yes.

AdamM68, 2021-04-20:
still there is NO answer how to power the board/display other way than the USB-C connector.
I'm also looking for the easy answer - solution : how to power this device (not by USB-C) ? 
- which pin should I use with 5v to make it work the same way as it works during USB-C powering ?
-

tadam777, 2021-04-30:
I'm powering it by the 5V pin, without using a battery or USB and it works fine.

itamardv, 2021-05-24:
Still without answer about how to use a battery and the back connector to power this thing?

dan-glass, 2021-05-24:
I connected the supplied power connector cable to a 4.7v lithium battery, and charge via the USB port. Works reliably and performs battery management.

perlix, 2022-07-26:
Two T-Display boards that I bought in early 2020 cannot be powered over their 5V pin. Three recently bought ones can, so there may have been a revision.

Shohreh, 2022-11-26:
I know nothing about electronics.

The TTGO has no on/off switch for the battery.

What compact switch could I use, so that I don't have to plug/unplug the JST cable each time?

Thank you.

flashtel, 2022-12-09:
I tried powering via the 5vdc and one of these 

[image]

The GPS has 12 satellites but keeps loosing fix.

I plug the USB in disconnect the 5v drive, rock solid fix. Bring it in side it get's down to 4 sats and has rock solid fix. 

It's also show batt volts when there is no battery attached when you drive it via the via 5 volt pin. 

I'm going to either modify a USB plug or try and get something onto the back of the USB.

Thanks

codiak6335, 2023-01-03:
5v on the 5v pin just fried the board.

While I don't mind paying for learning... the time it takes to replace the boards... ARGH

Has any got a reliable solution?  can I power the 3.3v with 5v safely?

flashtel, 2023-01-03:
Hi, 

I cooked mine also. 

The only solution I can come up with is power it directly through the USB port. 

I tried to pick up the line behind the USB connector but it was too fine for my soldering iron. 

https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.1.pdf 

I would buy a mini usb connector and power it through there. 

It's going to be ugly with a usb connector sticking out but I wish I had done that, rather than where we are both now. 

After I fried it, I tried to replace the AXP192 chip, I think the problem is down stream from that, it didn't work. 

I then tried a T-Motion board, which worked really well for about 7-8hours then it lost GPS fix, restarted wouldn't work. 

Waited overnight, restarted it worked for another 7 hours and now the GPS is coming up as not recognised on boot. 

I checked all the voltage test points and the 5v supply for noise/spikes looks good. 

I have a badge/t-echo as well, it works really well with low GPS signal, I think the quectel L73K GPS is better than the Ublox M8N, after about 30 hours the e-paper screen has gone U/S. 

3 out of 3 hardware faults. 

I spent a lot of time on this, not sure I want to spend anymore. 

Thanks

codiak6335, 2023-01-03:
I can add a 5v to 3.3v regulator inline... but seems a waste

Esantacruz01, 2023-02-15:
I had the same problem to power the board/display other way than the USB-C connector. Some boards doesn't work powering 5V from external. The board just works from USB-C connector.
Solution: To power from battery or from external source you need to add these two lines at beginning of your coding setup() function (Arduino IDE). Lines are pinmode(15,OUTPUT); digitalWrite(15,1);
Thios will enables the pin for external power. 
I hope that helps.

elpidiovaldez, 2023-03-02:
Is it not possible to power the board from 5V connected via the JST ? As far as I can see the charger chip (TP4054) allows battery voltage in range -0.3-7V, so 5V should not be a problem.  It is not going to try to charge a battery at that voltage, even if USB power is connected.  The 3.3V regulator allows input up to 6V, so that should continue to work.  I believe the other circuitry should prevent the USB supply fighting the battery input.  Am I missing anything ?  Has anyone tried it ?

Esantacruz01, 2023-03-02:
> Is it not possible to power the board from 5V connected via the JST ? As far as I can see the charger chip (TP4054) allows battery voltage in range -0.3-7V, so 5V should not be a problem. It is not [...]

Hi Valdez. 
I am unable to power up from the JST too.
Looks like this issue is related to some version revision of these boards....Some works, some doesn't. My lot of 10 boards just powers up from the USB-C. That's why you need to add the lines above in your coding...

elpidiovaldez, 2023-03-03:
What does GPIO15 do ?  On my schematic it does not appear to be connected to anything.

jessedmatlock, 2023-04-22:
> To clarify the different power options :
> 
> 1. power via USB without battery
> 2. power via USB with a battery : the battery will get charged, and will be used when USB is disconnected
> 3. power via the +5V. It won't charge the battery. This should be done without battery only : the 5V pin is connected directly to the battery (via Q5) as soon as USB is plugged in.
> 
> [image]
> 
> 4. power via the VBUS (which is the 5V pin of USB). This is the best approach, as this will charge the battery, and if no power is applied on VBUS then the battery will power the board. As can be seen on the schematic, this is the only way to charge the battery, the +5V or +3.3V pins won't do it. The problem is that VBUS isn't directly accessible on a pin. Of course VBUS is the same trace as the USB power, so ideally this should be used without the USB, even if it would probably work.
>    [image]

Thank you for the clarification!

erazor83, 2023-09-04:
Thanks for this issue. I also was wondering how to power via external 5V.

Using the battery input did not work for me. Fortunately I also didn't fried my board cos I noticed it's getting warm when I connect to the 5V PIN.

I connected an unused NC PIN directly to PIN 4 of the charger, see attached photos.
[image]
[image]

Achref-omrani, 2024-02-05:
> Powering it by the JST connector is working (you can put a 18650 bat to test). I could power it by USB-C with some voltage range.
> 
> Powering it by the 5V should work. Just try it.

Does this apply for both lipo and lion batteries ? It only needs to be 3.7v that's it the type do not matter ?