Xinyuan-LilyGO/TTGO-T-Display #28: Eagle library

mhannebert, 2020-04-13:
Did anyone create the library for use of this board in an Eagle project?

If not, I'm going to do it and share ;)

flyandi, 2020-04-15:
I made one a while ago. Attached. Also added as PR.

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/pull/29

[TTGO-TDisplay.lbr.zip](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/files/4478957/TTGO-TDisplay.lbr.zip)

flyandi, 2020-04-15:
This footprint has been tested in a real PCB btw. So it works.

mhannebert, 2020-04-15:
Thanks @flyandi 
I may work a little on it - and will post it back here

May I ask? How did you power up the device? 3.3V on one of the 3V3 pins?

flyandi, 2020-04-15:
Through the USB-C connector or the battery connector on the backside.

mhannebert, 2020-04-20:
Here is the library I created based on flyandi's.
I did not tested it yet but I'm pretty sure that it is correct.

[TTGO T-Display.lbr.zip](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/files/4504282/TTGO.T-Display.lbr.zip)

Would be nice to put a message if you used it.

M

flyandi, 2020-04-20:
Nice work. I will be updating my lib with this as well.

joanbono, 2020-04-26:
Just for the record, I used SnapEDA to convert the library from Eagle to KiCAD.

Thanks a lot guys (@flyandi @mhannebert )!

Attached:

[MODULE_TTGO_T-DISPLAY.zip](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/files/4535701/MODULE_TTGO_T-DISPLAY.zip)