Xinyuan-LilyGO/TTGO-T-Display #32: Which PINS are free to use for another SPI device?

minkione, 2020-04-18:
Hi, 
Which pins are still free from the TFT in order to be connected to another SPI device?

JoeBarouneD, 2020-04-20:
#14 See [this answer](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/14#issuecomment-584471342).

minkione, 2020-04-20:
thanks! Will give it a try.
Quick question, since I will need also 2 more free spare GPIOs...
Which are the GPIOs that can still be used overall?

JoeBarouneD, 2020-04-20:
You can use almost all other GPIOs.
Have a look at the Pinmap. Many GPIOs are free.

minkione, 2020-04-20:
I am asking because respect the usual ESP32 the pinouts for SPI doesn't look the same.
[image]

Versus the T-Display... (marked in red the pins in scope)
[image]

minkione, 2020-04-20:
Here I am comparing the ESP32 pinout and the pinouts from the T-Display schematics.
Now makes everything more clear. Though a doubt still remain regarding VSPI and HSPI... how do you enable the VirtualSPI (VSPI)? Just connect the device to those Pins and then declare in the sketch and everything works?

[image]

flyandi, 2020-07-29:
HSPI is only on 27,26,25 (MOSI, MISO, SCLK)  - at least from my experimentation .. CS pin can be anything.
VSPI can be on any free pin..

siriokds, 2021-06-08:
I used this for sd card access.

SPIClass spiSD(HSPI);

pinMode(2, OUTPUT);
spiSD.begin(13,12,15,2);       //SCK,MISO,MOSI,SS

raomin, 2022-02-08:
I wanted to connect a second display (ILI9341) on a secondary SPI.
While 27,26,25 (MOSI, MISO, SCLK) and 33 CS is ok, I found out that the pins 37 38 39 do not work for DC (Data command), RST (Reset) (or CS...)
Pins on the other side (17,2,15,12) works for DC, RST etc.

I ended up with:
```c
#define ILI9341_DRIVER

#define USE_HSPI_PORT
#define TFT_MISO  27  
#define TFT_MOSI  26  
#define TFT_SCLK   25 
#define TFT_CS    2

#define TFT_DC   33  // Data Command control pin
#define TFT_RST  32  // Reset pin (could connect to NodeMCU RST, see next line)

```

lockie555, 2023-07-23:
I am having an issue with using 3 devices (SD, 9341TFT, & RM3100) on SPI on a TTGO-T4v1.3.
My code works fine on a STM32 Nucleo152 but I cannot get the SPI bus to work on the T4. As the SD and TFT are built in on the T4, I only have access to the jumper pins and I chose SCLK-32; MISO -35; MOSI-34; CS-33 for my RM3100 magnetometer sensor.
The documentation re SPI on the T4 is a bit lacking, and I assume the SD and the TFT are both on default VSPI. I have tried the RM3100 as a HSPI but got nowhere; nor does it work as a VSPI. The TTGO-T4 pinout diagrams show the SD using SCLK -14, MISO-02, MOSI-15, CS-13 which are not the HSPI defaults and the TFT using SCLK -18, MISO-12, MOSI-23, CS-27 which are not VSPI defaults. 
I tried LewisXHE's code for SD with the TTGO-T display and had no luck with that either. I am not sure whether the problem is a lack of understanding by me. I learn by example and I haven't been able to find any examples of external sensors using SPI on T4. Any ideas?