Xinyuan-LilyGO/TTGO-T-Display #35: How to measure battery voltage?

Sakvojage, 2020-05-12:
```
uint16_t v1 = analogRead(34);
uint16_t v2 = analogRead(14);

float battery_voltage = ((float)v1 / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);
float other_voltage = ((float)v2 / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);

String voltage = String(battery_voltage) + "V " + String(other_voltage) + "V";
tft.setTextDatum(BR_DATUM);
tft.drawString(voltage, tft.width(), tft.height() - 5);
```
This gives me only USB power and constant 7.2V for pin 14 (whatever that is).
How can I track LiPo battery voltage/capacity?

JoeBarouneD, 2020-05-13:
Hello, i have tried to monitor voltage of an input solar panel to charge battery.

When there is an input, the GPIO34 displays the charge voltage (solar panel voltage).
When there is no input (only the battery), the GPIO34 displays the battery voltage (and you can check the capacity).

I don't know why you have a constant 7.2v on GPIO14, i will test and report it.

Bighoneypot, 2020-05-21:
Idem issu Sakvojage, only 7.2 V
I Have try pin 35

Bighoneypot, 2020-05-21:
> ```
> uint16_t v1 = analogRead(34);
> uint16_t v2 = analogRead(14);
>
> float battery_voltage = ((float)v1 / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);
> float other_voltage = ((float)v2 / 4095.0) * 2.0 * 3.3 * [...]

Try This
`#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip
#include <SPI.h>
#include <Arduino.h>
const uint8_t blue = 2;
const uint8_t vbatPin = 34;
#define TFT_GREY 0x5AEB // New colour
TFT_eSPI tft = TFT_eSPI();  // Invoke library
float VBAT= 0;  // battery voltage from ESP32 ADC read

void setup(void) {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(3);
  pinMode(blue, OUTPUT);
  pinMode(vbatPin, INPUT);
  //pinMode (16, OUTPUT);

}

void loop() {

  VBAT = (float)(analogRead(vbatPin)) * 3600 / 4095 * 2; 
  Serial.println("Vbat = ");
  Serial.print(VBAT); Serial.println(" Volts");
  digitalWrite(blue, 0);
  delay(2000);
  Serial.println("Vbat2 = ");
  Serial.print(VBAT); Serial.println(" Volts");
  digitalWrite(blue, 1);
  delay(5000);
  tft.fillScreen(TFT_BLACK);
  tft.setCursor(53, 50, 22);
  tft.setTextColor(TFT_YELLOW,TFT_BLACK);tft.setTextFont(4);
  tft.println(VBAT);
  delay(3000);`

Bighoneypot, 2020-05-21:
added VBATOK
#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip
#include <SPI.h>
#include <Arduino.h>
const uint8_t blue = 2;
const uint8_t vbatPin = 34;
#define TFT_GREY 0x5AEB // New colour
TFT_eSPI tft = TFT_eSPI();  // Invoke library
float VBAT= 0;  // battery voltage from ESP32 ADC read

void setup(void) {
  Serial.begin(115200);
  tft.init();
  tft.setRotation(3);
  pinMode(blue, OUTPUT);
  pinMode(vbatPin, INPUT);

}

void loop() {

  VBAT = (float)(analogRead(vbatPin)) * 3600 / 4095 * 2; 
  float VBATOK = VBAT / 1000;
  Serial.println("Vbat = ");
  Serial.print(VBATOK); Serial.println(" Volts");
  digitalWrite(blue, 0);
  delay(2000);
  Serial.println("Vbat2 = ");
  Serial.print(VBATOK); Serial.println(" Volts");
  digitalWrite(blue, 1);
  delay(5000);
  tft.fillScreen(TFT_BLACK);
  tft.setCursor(53, 50, 22);
  tft.setTextColor(TFT_YELLOW,TFT_BLACK);tft.setTextFont(4);
  tft.println(VBATOK);
  delay(3000);

Bighoneypot, 2020-05-26:
close #35

hmm-still-thinking, 2020-10-12:
> ```
> uint16_t v1 = analogRead(34);
> uint16_t v2 = analogRead(14);
>
> float battery_voltage = ((float)v1 / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);
> float other_voltage = ((float)v2 / 4095.0) * 2.0 * 3.3 * [...]

Can you explain why do we have to mutiply with 2 and what does  (1100 / 1000.0) mean? Thanks

caseyjmorton, 2020-10-21:
Not sure about the multiply 2 yet but the 1100/1000.0 is just lazy copying from the factory test example.  the 1100 is the default vRef.  It looks like this code was taken from https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TFT_eSPI/examples/FactoryTest/FactoryTest.ino and vRef was just replaced with the default 1100.

ketan, 2020-11-19:
Did anyone find a solution to this?

snst, 2020-12-09:
First I also read 7.2 volts when using the adc functions from the micropython port.
But then I used adc.atten(ADC.ATTN_11DB) to change the attenuation and then I read around 3.9 volts.
https://docs.micropython.org/en/latest/esp32/quickref.html#adc-analog-to-digital-conversion

andraz213, 2020-12-22:
`float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);`

So 1100 is actually 1.1V in mV. 1.1V is the internal reference voltage that readings get compared to.

Then `2.0 * 3.0` are voltage divider parameters. You can't measure 5V on an ADC with a 1.1V reference voltage you can only measure from 0V to 1.1V. That's why they used voltage dividers to extend the range to `2.0 * 3.3 * 1.1 = 7.26V`.
You actually have to set the reference voltage by setting the right attenuation `adc1_config_channel_atten(ADCx_CHANNEL_x, ADC_ATTEN_0db);` Something like this.

Then `v / 4095.0` is there because ESP32 has a 12bit ADC and 12bits is 4096 steps, so 0V is 0 and 1.1V is 4095. When you multiply that with 7.26, you get a range where 0 on ADC is 0.0V and 4095 is 7.26V.

andraz213, 2020-12-22:
They also built in an option to disable the voltage divider, so you have to set GPIO 14 to HIGH before measuring voltage and set it to LOW when not using it to preserve power. 

So there are two important things to do before reading the voltage:
- setting the attenuation to 0dB on the correct ADC and ADC channel
- setting the GPIO 14 to HIGH when measuring LiPo voltage

bbogdanmircea, 2020-12-22:
Isn't just better to post a code snippet that works for measuring battery voltage/ USB voltage and end this discussion?

andraz213, 2020-12-22:
I tested it.

This works:

#include <driver/adc.h>
#include "esp_adc_cal.h"

void setup() {
  Serial.begin(115200);
  esp_adc_cal_characteristics_t adc_chars;
  esp_adc_cal_value_t val_type = esp_adc_cal_characterize((adc_unit_t)ADC_UNIT_1, (adc_atten_t)ADC_ATTEN_DB_2_5, (adc_bits_width_t)ADC_WIDTH_BIT_12, 1100, &adc_chars);
  pinMode(14, OUTPUT);
}

void loop() {
  digitalWrite(14, HIGH);
  delay(1);
  float measurement = (float) analogRead(34);
  float battery_voltage = (measurement / 4095.0) * 7.26;
  digitalWrite(14, LOW);
  Serial.println(battery_voltage);
}

JoeBarouneD, 2020-12-22:
> They also built in an option to disable the voltage divider, so you have to set GPIO 14 to HIGH before measuring voltage and set it to LOW when not using it to preserve power.

From where did you get this information ?

andraz213, 2020-12-22:
[Quote from here:](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TFT_eSPI/examples/FactoryTest/FactoryTest.ino)

` ADC_EN is the ADC detection enable port
    If the USB port is used for power supply, it is turned on by default.
    If it is powered by battery, it needs to be set to high level`

SCIEN-00, 2021-04-06:
Does anyone know how the schematic of battery measurement part is designed? I am not able to measure precise voltage of battery when I use code samples posted above. It shows ~0,2V less than it actually is when measured (with dedicated voltmeter) from battery terminals. Also plugging in USB cable then value changes right away to 4,22V when measures with esp32. But when measuring on battery terminals it starts slowly charging. 🤔

andraz213, 2021-04-06:
The ADC in esp32 is not accurate and the resistors in voltage divider are also not accurate. You could try measuring the resistors and calibrating ADC, but it will never be as accurate as a multimeter.

seandepagnier, 2021-06-02:
How can I read the battery voltage when usb is plugged in?   I can't find a way to know when the battery is charged.

spattinson, 2021-08-21:
> How can I read the battery voltage when usb is plugged in? I can't find a way to know when the battery is charged.

The blue LED turns off. The battery voltage as measured by [this reply](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/35#issuecomment-749511521) while USB is plugged in ends up at about 4.78V when fully charged on my board

fgnievinski, 2024-01-15:
> I tested it.
> 
> This works:
> 
> #include <driver/adc.h> #include "esp_adc_cal.h"

indeed it works. you can plug in a battery and you'll see a realistic discharge curve
[image]

documentation:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/adc.html
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/adc.html#adc-calibration

PS: I believe this issue can be closed now.