Xinyuan-LilyGO/TTGO-T-Display #36: Board go to freeze after many "esp-light-sleep" calls

hpsaturn, 2020-05-23:
Hi, after many hours with `light-sleep` calls in the loop, the board go to freeze and the only action for turn on it is with reset button, in the log output with verbose level 5, I haven't anything, only it freezing without any error or panic core message, if I comment the `light-sleep` call, the device works for many days without any problem.

The complete code is [here](https://github.com/hpsaturn/CanAirIO) 

A brief is:

```java
void loop() {
    bmeLoop();                      // the BME680 sensor loop.
    wifiLoop();                     // check wifi or reconnect it.
    bleLoop();                      // notify data to phone or BLE device
    apiLoop();                      // CanAir.io API publication.
    influxDbLoop();                 // influxDB publication.
    otaLoop();                      // check for firmware updates.
    guiLoop();                      // gui ttf refresh methods.

    // save battery after phone disconnected:
    if (!bleIsConnected()) 
        espShallowSleep(5000);  <== F A I L    W I T H    I T
}

void espShallowSleep(int ms) {
    esp_sleep_enable_timer_wakeup(ms * 1000);
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
    esp_light_sleep_start();
}

```

lewisxhe, 2020-06-28:
@hpsaturn Please remove esp_sleep_pd_config (ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON); remove this line

hpsaturn, 2020-06-30:
@lewisxhe mmm but I can't remove this line because I need leave turned on the TFT. If I remove it, the peripherals go to off.