Xinyuan-LilyGO/TTGO-T-Display #38: SPI Pins (27,26, 25, 33) do NOT work with external SPI device

minkione, 2020-05-28:
I am trying to get working a CC1101 with my T-Display.
Someone suggested to device the following pins as SPI for teh connection between the T-Display and the NRF24...
MISO: 27
MOSI: 26
SCK: 25
CSN: 33
GDO0: 2
GDO2: 37
But it still doesn't work.

Anyone has some suggestions?

lewisxhe, 2020-06-28:
@minkione

As shown in the code below, did you specify SPI to initialize it?

```
#include "FS.h"
#include "SD.h"
#include <SPI.h>

SPIClass SPI1(HSPI);

 

#define MY_CS     15
#define MY_SCLK   25
#define MY_MISO 26
#define MY_MOSI 27
void setup()
{
 

    SPI1.begin(MY_SCLK, MY_MISO, MY_MOSI, MY_CS);

  //Assuming use of SPI SD card
    if (!SD.begin(MY_CS, SPI1)) {

        Serial.println("Card Mount Failed");

    } else {
        Serial.println("DEU CERTO INICIAR SD CARD");
    }
}
```

lewisxhe, 2020-06-29:
26,27 has been used, can not be used 26,27, you need to replace the pin

minkione, 2020-06-29:
In a SPI bus... there could be multiple slaves connected to the same MISO and MOSI and SCK.
What is important is to have a different CSN for each slave...
[image]

minkione, 2020-06-29:
That means that 26 and 27 must be used for both devices connected to the SPI bus.

lewisxhe, 2020-06-29:
Sorry, I thought you had another board, I made a mistake. I will briefly test it later.

minkione, 2020-06-29:
No problem :)
Actually after playing a bit I managed to get it work the CC1101.
But when I try to use BOTH CC1101 and the TFT Display... it fails badly to compile.

I uploaded here the sketch and the libraries needed, in case you wanna have a look. 
https://mega.nz/file/XEpXHb7A#rRXXcQ3D_pqLUPbAJQ455SL8r6hnwBaLgqQHPIcnbYM

P.S. Maybe is worth keep discussing in the other issue https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/14 and closing this one.

closed 2020-06-29 by minkione

lewisxhe, 2020-06-29:
I have tried using the SD card and everything is working, the code and README have been updated
@minkione

minkione, 2020-06-29:
Thanks anyway. I already opted for a MEGA2560 with a 3.2" TFT....
That had proper documentation  a month ago...