Xinyuan-LilyGO/TTGO-T-Display #40: A fatal error occurred: MD5 of file does not match data in flash! Help!!

juanjogarciagarcia, 2020-06-03:
I'm having this problem when i try to program my TTGO-T-Display.
I dont know how to fix.
Can you help me?

Arduino:1.8.9 (Windows 10), Tarjeta:"ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

El Sketch usa 214633 bytes (16%) del espacio de almacenamiento de programa. El máximo es 1310720 bytes.
Las variables Globales usan 15460 bytes (4%) de la memoria dinámica, dejando 312220 bytes para las variables locales. El máximo es 327680 bytes.
esptool.py v2.6
Serial port COM6
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:6f:28:99:99:d8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4681.2 kbit/s)...
File  md5: e6327541e2dc394ca2c3b3280ac0f39f
Flash md5: 0829f71740aab1ab98b33eae21dee122
MD5 of 0xFF is 84d04c9d6cc8ef35bf825d51a5277699

A fatal error occurred: MD5 of file does not match data in flash!
A fatal error occurred: MD5 of file does not match data in flash!

lewisxhe, 2020-06-28:
@juanjogarciagarcia  This situation is generally flash damage

Pelectrome, 2022-10-21:
Same problems for ttgo t watch v3, i test many solution but nothing works, hope to fine solution here