Xinyuan-LilyGO/TTGO-T-Display #42: PIN14 / 80MHZ SPI

flyandi, 2020-08-13:
So I am looking at the schematics and for some unknown reason the only pin to make 80MHZ IOMUX SPI possible is taken away for voltage measurement.

Is there any particular reason for that or is that something that LilyGo potential could change so we could use 80MHZ SPI .. for radical fast reading / writing speeds on SPI?

[image]

Thanks,

inoperable, 2020-08-13:
You can use pins 25, 26 and 15, for example;
```
#define CLK_PIN    25
#define MOSI_PIN  26 
#define MISO_PIN  27 
#define CS_PIN      15
```

CS can be remapped to (almost) any other not used GPIO.
Im running the board with 2 * 4 in 1:8x8 Max7219 DotMatrix modules in a chained config,
aside from connecting an extra power supply to run the DotMatrix (the drain too much the ESP32 pins)
the board seems to be working fine - allthough I didn't initialized the built-in OLED, but this should work
with it as well (gonna update as soon as I put a test of the OLED & MX7219 together)

flyandi, 2020-08-13:
But those are not IOMUX pins so you are limited to max 40mhz.