Xinyuan-LilyGO/TTGO-T-Display #43: Connect Second Display

j45p41, 2020-08-26:
Hi,

I have a project that requires two displays.

I am using the following mappings and using ILI9488 display :

#define TFT_MISO 19 // (leave TFT SDO disconnected if other SPI devices share MISO)
#define TFT_MOSI 26 //23
#define TFT_SCLK 17 //18
#define TFT_CS    15  // Chip select control pin
#define TFT_DC    33 // 2  // Data Command control pin
#define TFT_RST   27 //4  // Reset pin (could connect to RST pin)

Can you please help me use the second display? Both displays dont have to be on at the same time. EG integrated small display can be turned off but if both can be on at the same time then that would make life easier.

Thanks!

j45p41, 2020-08-26:
ok managed to get it working!

https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/42#issuecomment-673471021

next question is - where do i connect the touch pins please? :)

thefrog95, 2022-05-24:
could you elaborate how you got it to work?