Xinyuan-LilyGO/TTGO-T-Display #45: Control Backlight intensity

arduibag, 2020-10-06:
Do you have a feature to control the TFT backlight intensity? (with PWN management)

AReubens, 2020-12-03:
Yes, I was looking for this too, check out line 118 of User_Setup.h (and the comments before it):
```
#define TFT_BL   32            // LED back-light control pin
```

Then using the PWM controller on the esp you can change the backlight brightness:

```
ledcSetup(ledChannel, freq, resolution); // 0-15, 5000, 8
ledcAttachPin(TFT_BL, ledChannel); // TFT_BL, 0 - 15
ledcWrite(ledChannel, brightness); // 0-15, 0-255 (with 8 bit resolution)
```

Check out this [PWM on ESP32 tutorial](https://randomnerdtutorials.com/esp32-pwm-arduino-ide/) for more info

arduibag, 2020-12-03:
Thanks a lot!
For the frequency, it says "For an LED, a frequency of 5000 Hz is fine to use." but how is it calculated? Do you know?

rin67630, 2021-03-15:
Hi @AReubens,

Why do you recommend to check out line 118 of User_Setup.h?
TFT_BL is defined elsewhere it works without that.

rin67630, 2021-03-15:
Due to the specificity of the ESP32 to have 16 assignable  PWM controller channel, the user must define a channel to use.
The library should not use a channel on its own.

capedra, 2021-10-03:
On the file "Setup25_TTGO_T_Display.h", we can see that TFT_BL pin uses GPIO4. So, on User_Setup_Select.h file, we uncomment the following line:
```#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT```
 And after that, it's just a matter of controlling the brightness, which basically is:
```
pinMode(TFT_BL, OUTPUT);
ledcSetup(0, 5000, 8); // 0-15, 5000, 8
ledcAttachPin(TFT_BL, 0); // TFT_BL, 0 - 15
ledcWrite(0, 1); // 0-15, 0-255 (with 8 bit resolution); 0=totally dark;255=totally shiny
```

rin67630, 2021-10-04:
> For the frequency, it says "For an LED, a frequency of 5000 Hz is fine to use." but how is it calculated? Do you know?
Everything above 100Hz is fine.

nOfOKUz, 2023-05-14:
Hi,
Thanks for that tricks, that was what I was looking for.
Thought I am not seing any intensity changes on my TTGO screen going from 0 to 255. Is there something I missed? Is it working for you.
TT

hasaranga, 2024-09-05:
Found the issue. you need to set brightness after tft.init() function call.

  pinMode(4, OUTPUT);
  analogWrite(4, brightnessLevel);