Xinyuan-LilyGO/TTGO-T-Display #47: Implementing UART connection

DevilRulerEx, 2020-10-18:
Hello, I'm new to this.

With Arduino UNO, there are TX and RX pins for me to establish UART connection.

Which pins can I use to for TX and RX? Thanks.

DevilRulerEx, 2020-10-18:
Found the solution to my own issue,

This is a comment from banggood that has helped me:

A "standard" ESP32 has more pins, the default pins are 3 (Rx) and 1 (Tx) for Serial(0), 9 (Rx) and 10 (Tx) for Serial1, and 16 (Rx) and 17 (Tx) for Serial2. As you can see, with the limited number of pins on this TTGO board, none of them are there. But the good news is that the "Serial.begin" function has an option to change the pins! By using "Serial2.begin(9600, SERIAL_8N1, 25, 26);" you have the Serial2 port on pins 25 (Rx) and 26 (Tx). I tested this today, it's working like a charm!. It's NOT working on pins 37 and 38 (and maybe other pins), but it IS working on pins 25, 26, 27 !!!

tadam777, 2021-01-31:
Notice that GPIO pins 34 to 39 are input only, so they can be used for RX but not for TX.

henriquegmatos, 2021-12-16:
Please, There is a default pins to Serial2 or Serial1 on this board?

I´m try to use it as MIDI controller using MIDI.h on IDE Arduino. However I dont get define this pins (25, 26 or 27) to Serial to be used by MIDI.h library.

I try

`struct Serial2MIDISettings : public midi::DefaultSettings
{
  static const long BaudRate = 31250;
  static const int8_t RxPin  = 25;
  static const int8_t TxPin  = 26;

};

MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial2, MIDI, Serial2MIDISettings);`

but not works. Anybody could help me?

Thanks

rigr, 2023-02-12:
I'm facing similar problems using the TTGO-T Display board when using MIDI, WiFi and the display. I can not use the display at the same time I use MIDI. The following part works (I know, defining Serial2 is reduntant) but the display stops when it is called.
If using serial1 the board crashes. I'd love to use ths board for a MIDI-application WITH the display. I can only use it to send notes if I use AplleMIDI, but I'd love to use DIN-MIDi.

Anybody out there with a hint?

I also use those libraries:
```
#include <SPI.h>
#include <TFT_eSPI.h>
// 
TFT_eSPI tft = TFT_eSPI();
```

Andf this is the code:

```
struct SerialMIDISettings : public midi::DefaultSettings {
  static const long BaudRate = 31250;
  static const int8_t RxPin = 13;
  static const int8_t TxPin = 22;
};
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial2, MIDI1, SerialMIDISettings);
// MIDI_CREATE_INSTANCE(SoftwareSerial, midiSerial, MIDI);
void setup_MIDI() {
  Serial.println("setup_MIDI debug 00");
  Serial2.begin(31250, SERIAL_8N1, 13, 22);
  // midiSerial.begin(31250); // MIDI baud rate
  // DIN_MIDI.begin(MIDI_CHANNEL_OMNI);

   MIDI1.begin(MIDI_CHANNEL_OMNI); // Listen to all channels  
  
  // MIDI.turnThruOff(); // Disable MIDI thru
  //
  // Send MIDI on message
  Serial.println("setup_MIDI debug 01");
  MIDI1.sendNoteOn(60, 127, 1);  // Note On, C4, velocity 127, channel 1
  delay(500);
  // Send MIDI off message
  MIDI1.sendNoteOn(60, 0, 1);  // Note Off, C4, velocity 127, channel 1
  Serial.println("setup_MIDI debug 02");
}
```

rigr, 2023-02-12:
Just used Serial and that works. So it seems the board (or one of the display libraries) use Serial1 and/or Serial2.
Need to check out how to get the debug Serial-output to a different port.

This works for display AND DIN-serial-output:

To keep my code (more or less) clean I set a link to 
setup_MIDI();
inside 
void setup(){}

```
struct SerialMIDISettings : public midi::DefaultSettings {
  static const long BaudRate = 31250;
  static const int8_t RxPin = 13;
  static const int8_t TxPin = 22;
};
MIDI_CREATE_CUSTOM_INSTANCE(HardwareSerial, Serial, MIDI1, SerialMIDISettings);
// MIDI_CREATE_INSTANCE(SoftwareSerial, midiSerial, MIDI);
void setup_MIDI() {
  Serial.println("setup_MIDI debug 00");
  Serial.begin(31250, SERIAL_8N1, 13, 22);
  // midiSerial.begin(31250); // MIDI baud rate
  // DIN_MIDI.begin(MIDI_CHANNEL_OMNI);

   MIDI1.begin(MIDI_CHANNEL_OMNI); // Listen to all channels  /////////////////////////  will cause guru medidation
  
  // MIDI.turnThruOff(); // Disable MIDI thru
  //
  // Send MIDI message
  Serial.println("setup_MIDI debug 01");
  MIDI1.sendNoteOn(60, 127, 1);  // Note On, C4, velocity 127, channel 1
  delay(500);
  // Send MIDI message
  MIDI1.sendNoteOn(60, 0, 1);  // Note Off, C4, velocity 127, channel 1
  Serial.println("setup_MIDI debug 02");
}
```

navivfr, 2023-11-30:
Hello,

I get a TTGO LORA32, and has output pin for UART there is only GPIO 1 & 3 (used for USB serial).
 I need to add a SIM800 (GPRS).
I tried using GPIO 12 & 13... without any success. _(Serial2.begin(115200, SERIAL_8N1, SERIAL2_TX, SERIAL2_RX);_
Does someone know how to use UART with the board ?

thanks

tadam777, 2023-12-01:
Hello, it seems you reversed the TX and RX in your Serial2.begin command.

It should be SerialPort2.begin (BaudRate, SerialMode, **RX_pin, TX_pin**).

You'll find a tutorial on using the serial ports, including using alternative pins here : https://microcontrollerslab.com/esp32-uart-communication-pins-example/