Xinyuan-LilyGO/TTGO-T-Display #48: Pin 36, 37, 38, 39 capacitors

ddosoff, 2020-10-22:
Hello LiLyGO,

Today discovered that I can't use SPI_MISO on GPIO 38 and UART RX on GPIO 39 simultaneously.
I checked https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf and found capacitors between 39, 38 and 37, 36 GPIOs.

[image]

Please describe what purpose of this capacitors?

ddosoff, 2020-10-22:
Can I remove them from PCB?

bradanlane, 2020-12-06:
Thanks @ddosoff - I didn't catch this on my project. I've gone ahead and removed those CAPs.

I hope we get an answer as to their intended function.

tadam777, 2021-01-31:
There's no information about their role in the ESP32 datasheet, but they've got a short description in the ESP32 hardware design guidelines :
[image]

JFPayeur, 2022-09-19:
My guess is these have a role in touch sensitive switch .. But not shure how

tadam777, 2022-09-19:
I found an article about these capacitors, they're related to an LNA (low noise amplifier) that's actually included on the ESP32, but didn't work well so it was simply removed from the documentation, even if it still exists in the chip.

Someone took the version 1 of the documentation (where it was still described) to find out how to use it.

Full story here : https://github.com/krzychb/esp32-lna