Xinyuan-LilyGO/TTGO-T-Display #49: How to turn off the device when powered through battery?

msurguy, 2020-11-16:
Thank you for this display / device. It's very convenient and easy to use!

I was wondering how I can turn off the device completely when powered through battery? 
Is there a way to automatically shut off when battery voltage reaches some level? 

In the worst case I can put mechanical on / off switch, but that requires me to redo the wiring from the batteries. Any help on this would be appreciated!

lewisxhe, 2021-01-14:
You can't physically shut down using electromagnetic, you can only set sleep, if you need to physically shut down, you only need to connect a switch between the battery and the board.

Batlapin, 2021-03-13:
It is possible to significantly reduce power consumption rewiring with come extra components (controlling LDO's EN pin) but you must decide how are you going to power module up after shutting it down.