Xinyuan-LilyGO/TTGO-T-Display #50: Centering Text on Display

thevalo, 2020-11-22:
Hello, I've setup screen width and size (240x135) in UserSetup.h and when I use setTextDatum as MC_DATUM it still doesn't center the text in the screen. Are there any function to center text or what should be the setCursor parameters so that text is always centered?

mikeprice99, 2020-11-27:
It seems to work perfectly for me. 
Can you give an example of the code that doesn't work (ie what command are you giving to write your text)?

thevalo, 2020-12-03:
Hello, this is the code:
```
#define FF18 &FreeSans12pt7b
#define GFXFF 1
 tft.setTextDatum(MC_DATUM);
 tft.setTextColor(TFT_WHITE, TFT_BLACK);
 tft.fillScreen(TFT_MAGENTA);
 tft.setFreeFont(FF18);    
 tft.drawString("Yellowtail 32", 160, 60, GFXFF);
```
The text "Yellowtail 32" is centered vertically, but not horizontally, so it's slightly spaced to the right.

kakaki, 2021-12-29:
Last line should be: 
tft.drawString("Yellowtail 32", **120**, 60, GFXFF);
change 160 to 120, (screen size is 240x135) so 120 is in middle of screen not 160.

closed 2022-01-28 by thevalo