Xinyuan-LilyGO/TTGO-T-Display #52: Is my TTGO-T-Display dead or is there a magic trick?

DavidBertet, 2020-12-19:
I just received a TTGO-T-Display. It wasn't doing anything when I plugged it (which isn't normal from what I read), so I decided to push a sample.

Whatever I build from this repo, I always have a part of the screen black, a part green, and a part kind of ugly but correct.

[image]

[image]

Is there a magic trick I missed? Is my screen dead? Could it be another component?

Two things I noticed too:
- the blue LED is not fully bright, it blinks a little bit
- the USB C isn't reversible. it works only in one way, it doesn't power up on the other way
Is that normal?

Thanks!

lewisxhe, 2021-01-14:
Do you mean using the examples provided by this repository to show what it looks like in the picture? It may be that the screen has been damaged.

The blue led light flashes, it means that the battery is not connected. If the battery is connected, it will indicate according to the charging status.

USB, what's wrong, I don't understand

JoeBarouneD, 2021-01-14:
I had same results when i pushed an outdated/bad configurated library.

Be sure to use to correctly install TFT_eSPI and declare the good I2C ports according the sheet.
Don't forget to customize the [User_Setup.h](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TFT_eSPI/User_Setup.h) if you use your own version of TFT_eSPI