Xinyuan-LilyGO/TTGO-T-Display #53: I2C Connection problem

GodSmithCS, 2020-12-26:
Hello, I'm new using ESP, but I'm trying to use ADS with the T-display and I can't make the connection. I am using the example of the I2C Scanner and I believe that the wires are correct according to the datasheet.

`
 #include <Wire.h>

 void setup() {
  Wire.begin();

  Serial.begin(9600);
  while (!Serial); // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
  }

void loop() {
  int nDevices = 0;

  Serial.println("Scanning...");

  for (byte address = 1; address < 127; ++address) {
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
      Serial.print("0");
      }
      Serial.print(address, HEX);
      Serial.println("  !");

      ++nDevices;
    } else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) {
      Serial.print("0");
      }
      Serial.println(address, HEX);
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  } else {
    Serial.println("done\n");
  }
  delay(5000); // Wait 5 seconds for next scan
}
`

[image]

lewisxhe, 2021-01-14:
Which pin do you use?
You can explicitly initialize the designated I2C pin,
```
Wire.begin(sda,scl)
```
Is your external pull-up resistor 110 ohms? You can use 4.7k or 10k for external pull-up.

tadam777, 2021-02-12:
This board already includes 10k resistors on SDA and SCL :

[image]

For some reason I'm not able to use the default I2C pin 22, but for example with pins SDA = 21 and SCL = 15 it works flawlessly.

Notice that pins 35 to 39 are input only so they can't be used for I2C.

#define I2C_SDA 21
#define I2C_SCL 15

void setup()
{
	Wire.begin (I2C_SDA, I2C_SCL);
	pinMode (I2C_SDA, INPUT_PULLUP);
	pinMode (I2C_SCL, INPUT_PULLUP);
}

FlyByPC, 2021-02-22:
SCL is defined as pin 23 (not 22) in the board definitions, despite what the pinout diagram says.
Unfortunately, Pin 23 isn't brought out to the pins.

Change the SCL pin definition in C:\Users\<user>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\variants\ttgo-t1\pins_arduino.h and it will work.

lewisxhe, 2021-03-11:
Except for the specified SDA SCL, everything else is the same. I can work normally. Please rule out the problem with the module, or you can try with an I2C device. 

```
#include <Wire.h>
#include <Arduino.h>

const uint8_t sda = 21;
const uint8_t scl = 22;

void setup()
{
    Serial.begin(9600);
    Wire.begin(sda, scl);
    while (!Serial); // Leonardo: wait for serial monitor
    Serial.println("\nI2C Scanner");
}

void loop()
{
    int nDevices = 0;
    Serial.println("Scanning...");
    for (byte address = 1; address < 127; ++address) {
        Wire.beginTransmission(address);
        byte error = Wire.endTransmission();

        if (error == 0) {
            Serial.print("I2C device found at address 0x");
            if (address < 16) {
                Serial.print("0");
            }
            Serial.print(address, HEX);
            Serial.println("  !");

            ++nDevices;
        } else if (error == 4) {
            Serial.print("Unknown error at address 0x");
            if (address < 16) {
                Serial.print("0");
            }
            Serial.println(address, HEX);
        }
    }
    if (nDevices == 0) {
        Serial.println("No I2C devices found\n");
    } else {
        Serial.println("done\n");
    }
    delay(5000); // Wait 5 seconds for next scan
}

```

m-stefanski, 2021-11-24:
> #define I2C_SDA 21
> #define I2C_SCL 15
> 
> void setup()
> {
> Wire.begin (I2C_SDA, I2C_SCL);
> pinMode (I2C_SDA, INPUT_PULLUP);
> pinMode (I2C_SCL, INPUT_PULLUP);
> }

when I am trying to do it this way, LCD screen shows garbage and I2C connection still doesn't work. 

[image]

Tested on PN532 with Adafruit library:

```

#define I2C_SDA 21
#define I2C_SCL 15

void initialize_PN532(){
    Serial.println("Initializing PN532");
    Wire.begin (PN532_SDA, PN532_SCL);
    pinMode (PN532_SDA, INPUT_PULLUP);
    pinMode (PN532_SCL, INPUT_PULLUP);
    nfc.begin();
    uint32_t versiondata = nfc.getFirmwareVersion();
    if (! versiondata) {
        Serial.print("Didn't find PN53x board");
        while (1); // halt
  }
}
```

This happens also when I used pins 26 / 27.

tadam777, 2021-12-11:
In your code you define the **I2C_SDA** and **I2C_SCL**, but then you write Wire.begin (**PN532_SDA**, **PN532_SCL**) so you're probably not using the pins 21 and 15.
Could it be that because of this you're using a pin which is connected to the TFT and this give the random pixels on the screen ?

Have a look at the example above, 
#define **I2C_SDA** 21
#define **I2C_SCL** 15

void setup()
{
Wire.begin (**I2C_SDA**, **I2C_SCL**);
pinMode (I2C_SDA, INPUT_PULLUP);
pinMode (I2C_SCL, INPUT_PULLUP);
}

This works for me, even without the two INPUT_PULLUP lines

summerfind, 2023-01-18:
In your code you define the **I2C_SDA** and **I2C_SCL**, but then you write Wire.begin (**PN532_SDA**, **PN532_SCL**) so you're probably not using the pins 21 and 15 that you defined first. Could it be that because of this you're sending SDA/SCL data to another pin which is connected to the TFT and this give the random pixels on the screen ?
> 
> Have a look at the example (from above) :
> 
> #define **I2C_SDA** 21 #define **I2C_SCL** 15
> 
> void setup() { Wire.begin (**I2C_SDA**, **I2C_SCL**); pinMode (I2C_SDA, INPUT_PULLUP); pinMode (I2C_SCL, INPUT_PULLUP); }
> 
> This works for me, even without the two INPUT_PULLUP lines.
> 
> I can also recommend this small sketch, which will detect any I2C device and give its address. It can be useful if you board can be configured for several addresses : https://playground.arduino.cc/Main/I2cScanner/

Hi all, is this too later?
can the config here be able to set other pin than 21/22, let's say 13/14?
Thanks