Xinyuan-LilyGO/TTGO-T-Display #56: 5V without USB from battery only?

bbogdanmircea, 2021-02-16:
Hello, I have a project where I need to power a device with minimum 4.5V. (CO2 Sensor)
But the 5V pin will drop to battery voltage when there is no USB Power.
Is there any way to have 5V from.the battery with this board?
Or I should buy a small battery boost DCDC converter that I connect in parallel to the battery?
Also VBAT and BAT are the same ?

Docteh, 2021-02-25:
Interesting question, looks like board doesn't have parts to boost voltage. 
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/schematic/ESP32-TFT(6-26).pdf

At the bottom of schematic they show where the "+5" pad is either battery or "VBUS" from USB. I only saw one reference to BAT, maybe they missed the letter?

Batlapin, 2021-03-13:
1) No boost from LiIon to 5V onboard. Extra boost circuitry should be used.
2) "1" at PWR_EN (IO14) shortens BAT to VBAT (and V3V, TFT communication pins pullups power, to VDD3V3)