Xinyuan-LilyGO/TTGO-T-Display #61: Which board to select on PlatformIO?

CS-Account, 2021-05-10:
I have been using ttgo-lora32-v1, and it has worked, but I figured it would make sense to ask, since I haven't seen a definitive answer and the TTGO T-Display doesn't match any of the board names available on platform IO directly.

Having this as part of the guide could be useful.

kgrozdanovski, 2022-01-08:
You can use the following settings to successfully use this board with PlatformIO:
```
platform = espressif32
board = esp32dev
```
The team at PlatformIO [has stubbornly been ignoring requests](https://github.com/platformio/platform-espressif32/issues/209) to support this board for 3 years now, so I wouldn't wait for an official listing.

tadam777, 2023-12-06:
Lilygo T-Display **S3** is now supported in PlatformIO out-of-the-box with "board = lilygo-t-display-s3" in platformio.ini
https://docs.platformio.org/en/latest/boards/espressif32/lilygo-t-display-s3.html

Note that the T-Display **non-S3** isn't supported in PlatformIO but a workaround is described here : https://github.com/JakubAndrysek/TTGO_T_Display/blob/master/TTGO_example/platformio.ini

mintisan, 2024-03-04:
speed shoud lower, if enrounter fatal-error-occurred-invalid-head-of-packet error

```
#upload_speed = 921600  #too higher
upload_speed = 460800
```