Xinyuan-LilyGO/TTGO-T-Display #62: Powering off

oculos, 2021-05-11:
Hi,

I wonder if there's a combination (like connecting EN to GND) to disable the 3.3V regulator and turn the TTGO off. I know I could use deep sleep, but I'd rather cut all power from the device. Is that possible?

Does the TTGO have a EN pin?

tadam777, 2021-05-25:
If you power from USB, there will always be power on the 3.3V
But I assume you're on battery, did you try the PWR_EN pin ? Its role isn't completely clear to me, but it looks like it commands a connection between the battery and the 3.3V regulator.

[image]

JFPayeur, 2022-09-19:
Set PWR_EN to 1 enable the battery voltage reading , but is consume a little tiny bit of power. It does nothing to power off the device

tadam777, 2022-09-19:
Yes, it's at least 50µA through R27 (and maybe as much through R28), which can seem very low, but it's relatively high in comparison to the ESP32 sleep modes (deep sleep 10µA, hibernation 2.5µA).

JFPayeur, 2022-09-20:
And It does power the "BAT" pin , 
And BAT power R6/R7 
[image]
In the same order off magnitude of power loss