Xinyuan-LilyGO/TTGO-T-Display #65: Deep sleep wake up only over Connect USB

Michaelmilz, 2021-06-06:
Hi, 

i have the Problem that my TTGO T-Display Board does not wake up when iam powering that only with the Battery. When i connect the USB cable to the Board, the wakeup over ext0, ext1 or externel Timer works without Problems. For the wakeup i use the Button that Connected to GPIO0. On The other Button (GPIO35) Same Issue. Powering over Battery i only can use Reset Button to bring the Board awake. 
Have someone the same issue or an idea what iam doing wrong ?

g-arais, 2021-09-23:
Same with me. The board only wake up by ext0 or ext1 on USB power. When on battery, only reseting. Any sugestions?

Michaelmilz, 2021-09-23:
I solve the problem myself. Because it was long ago I don't know exactly what Iam done. 
I figured out that the issue was in the example program. I excluded everything piece by piece until I found the failure code. It may be that the problem was in the analog Measurement of the battery voltage. Write if it helps.

roberbike, 2021-12-29:
Leaving this lines, it is possible to wake up with io34 , battery supply and usb disconnected.

    rtc_gpio_init(GPIO_NUM_14);
    rtc_gpio_set_direction(GPIO_NUM_14, RTC_GPIO_MODE_OUTPUT_ONLY);
	rtc_gpio_set_level(GPIO_NUM_14, 1);
    delay(500);

Jpipe001, 2022-05-04:
Hi,

I've got a TTGO T-Camera (the white version, with a PIR) which won't wake from deep sleep 
when powered via a 3.7V LiPo battery attached to the on-board battery connector.

I have written a sketch where the Camera goes to Sleep and then Wakes Up, either when PIR is triggered or after a wait period.
 
Sleep/Wake behaves completely normally when powered via USB.
 
But with the battery attached and not powered via USB:
If I let it sleep, then trigger a wake within a few seconds, it works.
If I let it sleep, leave it for ~10 seconds or so, it will not Wake up.

How do I configure this to Wake Up on Battery Power?

Any advice greatly appreciated!