Xinyuan-LilyGO/TTGO-T-Display #69: Cannot flash new TTGO-T-Display

cgreening, 2021-09-03:
I've received a new TTGO-T-Display and for some reason, it's not flashing:

I have another TTGO-T-Display board with a dead display that flashes without any problems.

```
esptool.py v3.1
Serial port /dev/cu.usbmodem527E0006811
Connecting.....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:45:34:64
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000)
A fatal error occurred: Failed to write to target RAM (result was 01070000)
```

This is what I see with my old board:

```
esptool.py v3.1
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c4:4f:33:7f:f5:b1
Uploading stub...
Running stub...
Stub running...
```

cgreening, 2021-09-04:
After installing some drivers on another issue this has progressed, but not much further:

```
esptool.py v3.1
Serial port /dev/cu.wchusbserial527E0006811
Connecting......
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:45:34:64
Uploading stub...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
```

cgreening, 2021-09-04:
The device also works perfectly on windows so it's definitely an issue with the Mac driver.

coryb, 2021-10-05:
I had the same issue, I think the key is `/dev/cu.SLAB_USBtoUART` which is created by the SiLabs Driver Disk package.  I installed this previously for a different esp32 devkit, and this drivers seems to conflict with the usb drivers necessary for the TTGO.  I had an application called "CP210xVCPDriver" which was installed by the SiLabs Driver Disk.  I moved that application to the "Trash" and was prompted to  uninstalled the drivers.  After that my TTGO started to work after I reconnected it to my mac.

raybutch, 2022-01-03:
I'm having the same problem. I am unable to flash the TTGO T display module. I am looking for a driver to load on my windows 7 computer. Does anyone know of a driver that will work in a windows?

fmgdias, 2022-01-29:
Connect Pin 2 to GND.

piperov, 2022-04-28:
Similar problem (I think) on a Chromebook. Working in the Linux container, I see a USB device showing up when I connect the TTGO-T-Display board, but no /dev/ttyXXX device gets created. 

Conclusion: the USB device on the board fails to be recognized/configured as a Serial Port. Probably a driver issue. 

The same procedure, repeated on the same computer, but when running Linux directly on the bare metal, sees a /dev/ttyACM0 created, and then I can program the TTGO-T-Display board, and interact with it in MicroPython (Thonny IDE) without any issue.

Something is fishy with the USB/Serial implementation on the TTGO-T-Display board though.
Could it be the same issue discussed in #47 ?
(https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/47)

mianos, 2022-06-06:
You can no longer flash with the CH9102 driver on MacOS Monterey. It sometimes works under Linux. The basic serial port works OK but flashing is hit and miss. I started the kex debugger and it seems to be passing misaligned structures to the API as if it was built with a pre 2018 MacOS driver SDK.

If you have a Windows machine it works 100%.
The WCH driver listed above works on some machines pre Monterey.

It is a bit of a fiasco but one way forward is to flash the additional OTA update configuration on a old Windows machine and then use a proper machine to flash it via wifi.