Xinyuan-LilyGO/TTGO-T-Display #73: Not flashing anymore

DaveStu, 2022-01-07:
Hi all. I've had this board for a couple of months with a project I'm working on with someone else.

I've been flashing .bin files with esphome flasher succesfully for the last few months, now suddenly it stops flashing.
The screen will start to flash several times while esphome flasher tries to connect, but it fails, then boots back to previous code and is able to work with last upload, just no new flashing?
I've been all over forums and internet looking for solutions, tried to hold reset, button 1 and/or 2.
Tried different tyep c data cables. I cannot disconnect components because it's all are sodlered in.

I'm now stuck with last code and cannot update our project.
Can I bypass USB and flash via pins, like esp01 with USB flasher?
This has now got me so frustrated.

Any clue anyone?  really strange, all of a sudden.  Thanks

DaveStu, 2022-01-08:
After reluctantly desoldering TTGO board, I was able to flash it again. Then I resoldered it back onto the PCB attached back component at a time, the line elevel converter for LED strips was causing some short or glitch.  I assume it has died or failed somehow.
I've now just left it disconnected and the TTGO flashes and functions fine.

fmgdias, 2022-01-29:
Connect Pin 2 to GND.