Xinyuan-LilyGO/TTGO-T-Display #80: Can this 18650 shield power a TTGO?

Shohreh, 2022-03-13:
Hello,

I'd like to power a TTGO board with a single 18650 battery and ditch the power bank I'm currently using.

Do you know if [this shield from EQV](https://www.aliexpress.com/item/4001113579534.html) and any 18650 battery is safe and does the job?

Thank you.

[image]

MrMcChip, 2022-03-13:
Which one of the many TTGO boards?
Does it have a dedicated DC battery connector onboard?

Shohreh, 2022-03-13:
I'm using [this one](https://www.aliexpress.com/item/33048962331.html).

I read its battery connector isn't reliable to charge a battery, which is why I'm looking at an extra part (this shield.)