Xinyuan-LilyGO/TTGO-T-Display #81: GPIO0 button and uploading

stevempotter, 2022-03-25:
Sometimes I can upload new code in Arduino without pressing any buttons, sometimes it aborts flashing code with "Fatal error. Timed out waiting for packet content". Holding down the GPIO0 button seems to fix it but is painful. Is there a way to get around this or make it more reliable to upload without pressing any buttons?

stevempotter, 2022-03-27:
Here is the error I keep getting, even sometimes when I am holding down the GPIO0 button. Any ideas how to prevent this?

```
esptool.py v3.1
Serial port COM23
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:11:ae:a5:f2:98
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x000b1fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

MrMcChip, 2022-03-28:
My first point of focus would be your programming USB cable.
 I have a dedicated 18-in cable that I know has consistent results with
multiple boards using both micro USB and USB-C (with add-on adapter).

stevempotter, 2022-03-28:
On your suggestion, I tried several cables, including some expensive thick well-shielded ones, and short ones. I still can only upload about half of the time. None of them fixed the problem. Sadly, I have gone back to an old DOIT Devkit ESP32 to avoid this problem.

MrMcChip, 2022-03-28:
A lot of troubleshooting is a process of elimination. Other variables to
consider are the board itself so if you have another board to work with or
test? The other simple thing to do is lower the board rate and see if that
shows you anything?

stevempotter, 2022-03-29:
Is there a way to understand this problem and measure something (e.g., with my oscilloscope) to see how to fix it?
I have tried several boards, and the problem is that they all sometimes work and sometimes give the timeout error. An intermittent problem is hard to track down by just trying random changes. My latest tests suggest that if the LilyGO ESP32 board has nothing attached but the USB-C cable, it always uploads. But I don't want to have to unplug the board from its circuits every time I upload new code.

stevempotter, 2022-03-29:
Here are my board settings:
[image]

MrMcChip, 2022-03-29:
Which UART bridge controller does your board have? Which operating platform
are you running?

stevempotter, 2022-03-30:
> Which UART bridge controller does your board have? Which operating platform
are you running?

I am running Arduino IDE 1.8.15 on Windows10. Here is a photo of the USB chip on the ESP32 that is giving me a hard time.

[image]

MrMcChip, 2022-03-30:
You have a Sil Lab CP 2104 which is the best available IMO. Get yourself a
datasheet.pdf for that bridge.  You should verify Windows is loading the
proper driver every time you plug in the USB with TTGO board attached
before doing your upload. If you get a timeout recycle the USB connector
with board attached. Windows driver utility should provide that Intel.

*USB sniffer and protocol analyzer for Windows would be my next move to
further troubleshoot USB port communications. This should prove more useful
than an oscilloscope at this point.*
*I operate on a Linux platform... I don't do windows!  So I can't make any
recommendations but there are options available on the web for this type of
utility. *

MrMcChip, 2022-03-30:
Which bridge controller chip is on your DOIT development ESP32 board? Are
they same same as the TTGO?

Also, I believe windows command line prompt will allow you to use the LSUSB
command... If it works in terminal mode/ command line you should see CP210x
listed

stevempotter, 2022-03-30:
> Which bridge controller chip is on your DOIT development ESP32 board? Are
they same same as the TTGO?

It looks the same as the SIL 2104 but has no markings whatsoever. Etched off? This DOIT board is probably a clone.

From Windows Device Manager:
Both boards invoke the "Silicon Labs CP210x USB to UART Bridge" driver when plugged in.

> Also, I believe windows command line prompt will allow you to use the LSUSB
> command...

"LSUSB" was not a recognized command at the Terminal command line.

I downloaded "Free Device Monitoring Studio" to look at the USB packets to and from the ESP32. But I have no idea what to look for, and have not used a packet sniffer before. Do you have any suggestions?

MrMcChip, 2022-03-30:
Look into USB device enumeration. It looks like the device and driver are
communicating when given a push button reset when needed. At this point I
would research your CP210X  driver version and see if there's an update
available from silicon labs? I believe your chip is date code 2015?
Again having no desire to give Bill Gates any of my money I'm not a Windows
fan