Xinyuan-LilyGO/TTGO-T-Display #83: External antenna

levycarneiro, 2022-04-03:
Is there a way to connect an external antenna to extend the range of Wifi and/or ESPNOW?

Thanks!

bobvc133, 2022-04-05:
Hope there is a way to install an external antenna, using a metal casing will shield the signal

LilyGO, 2022-04-06:
Hello, you can use the external antenna as shown in the picture
[image]

bobvc133, 2022-04-06:
TTGO T-DISPALY with CH340K chip can use IPEX external antenna
[image]

levycarneiro, 2022-04-06:
Found this tip on how to add an external antenna for ESP32 boards in general:

https://www.youtube.com/watch?v=-8hxk81H6QI

levycarneiro, 2022-04-07:
> Hello, you can use the external antenna as shown in the picture [image]

Can you provide more details? I can't quite discern what's being shown in the picture.

Thanks.

bobvc133, 2022-04-28:
> 您好，您可以使用如图所示的外置天线 [image]
图片放大还是看不太清楚，掉点处原来的的电阻是去掉了吗？还要把板载天线改成图片上那样吗？

bobvc133, 2022-04-29:
[image]
我的板子红圈处有一个电阻，是要去掉以后把天线焊接在黄圈两个位置对吧？

LilyGO, 2022-04-29:
对的

bobvc133, 2022-04-29:
> 对的

好的，非常感谢，手工焊接比较费劲，怕手残焊坏了，所以问的这么仔细