Xinyuan-LilyGO/TTGO-T-Display #92: [TTGO T8 ESP32-S2] Sample code fails compiling

Shohreh, 2022-07-09:
Hello,

I'm an Arduino + ESP newbie.

I installed the Arduino 2.0.0rc8 since I read the current stable release (1.8.19) doesn't support that TTGO board.

Then, I [added](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json) and selected that board + port, and [manually installed](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/tree/master/TFT_eSPI) the TFT_eSPI library.

But when I try to compile Arduino\libraries\TFT_eSPI\examples\FactoryTest\FactoryTest.ino, here is what I get:

```
In file included from C:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI\examples\FactoryTest\FactoryTest.ino:1:
c:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:302:21: error: 'VSPI' was not declared in this scope
     uint8_t  port = VSPI;
                     ^~~~
c:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI/TFT_eSPI.h:302:21: note: suggested alternative: 'SPI'
     uint8_t  port = VSPI;
                     ^~~~
                     SPI
C:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI\examples\FactoryTest\FactoryTest.ino: In function 'void setup()':
C:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI\examples\FactoryTest\FactoryTest.ino:212:90: error: 'ADC_WIDTH_BIT_12' was not declared in this scope
     esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);    //Check type of calibration value used to characterize ADC
                                                                                          ^~~~~~~~~~~~~~~~
C:\Users\dummy\Documents\Arduino\libraries\TFT_eSPI\examples\FactoryTest\FactoryTest.ino:212:90: note: suggested alternative: 'ADC_WIDTH_BIT_13'
     esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_11, ADC_WIDTH_BIT_12, 1100, &adc_chars);    //Check type of calibration value used to characterize ADC
                                                                                          ^~~~~~~~~~~~~~~~
                                                                                          ADC_WIDTH_BIT_13
Multiple libraries were found for "SD.h"
  Used: C:\Users\dummy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\SD
  Not used: C:\Users\dummy\Documents\Arduino\libraries\SD
exit status 1

Compilation error: 'ADC_WIDTH_BIT_12' was not declared in this scope
```

What should I try?

Thank you.

icedawn, 2023-01-25:
Did anyone find a workaround for this?  I'm testing the latest code and examples and still getting this error.

Thanks!

MichKonSher, 2023-01-25:
Same problem

Shohreh, 2023-01-25:
I put it on the back burner, but you might have more luck trying Bodmer's TFT_eSPI instead:

https://github.com/Bodmer/TFT_eSPI/issues/2022

icedawn, 2023-01-25:
Thank you Shoreh.  I was able to solve it with help from https://github.com/teastainGit.  Bottom line, I have T-Display-S3 boards, not TTGO boards -- so Bodmer's and the TTGO libraries weren't going to work until these libraries support the T-Display-S3 board.

Context:  there is a specific TFT_eSPI library provided by LilyGO for the T-Display-S3 at https://github.com/Xinyuan-LilyGO/TTGO-T-Display ... and you need to use the TFT_eSPI contained there.  Worked the first time.  Kudos go to teastainGit for figuring it out and being super helpful!

You can see his notes and our discussion at:  https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples/issues/3 for instructions.

Thanks!

Ed