Xinyuan-LilyGO/TTGO-T-Display #97: Lowest power consumption

JFPayeur, 2022-09-19:
Closing the screen 
    -->  digitalWrite(4, LOW); 

 Closing Wifi+BT 
      -->   WiFi.disconnect(true);  
      -->   WiFi.mode(WIFI_OFF);    // Switch WiFi off

Not reading the bat 
   -->  digitalWrite(POWER_EN_PIN, LOW); 
 
And going to deep sleep 

Best I can get is ~22ma consumption on the USB 

Seems the USB chip put itself to off when not powered, and this is fine 
So about 7.7 ma , when reading power from only the Battery ( most relevant metric) 

But still 7.7mA, is way to much considering the Esp32 could go in the uA range! 
Any other idea ?  

Seems to be a firmware available ,,,, 
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/tree/master/firmware

I've never done this type of upgrade 
Is this to fix this ?

JFPayeur, 2022-09-19:
Did the upload, With this tool 
https://www.espressif.com/en/support/download/other-tools

[image]
It was suprisigly simple 

The demo code can demonstrate a power down mode that I mesure down to 0.2mA in deep sleep. 
This is cool 
But my code still do 7.7mA 
Will have to investigate further

oculos, 2023-02-26:
Same problem here. 
I get 17-19mA when connected to the USB. I haven't measured it only connected to the battery.