Xinyuan-LilyGO/TTGO-T-Display #101: black screen

defaliz, 2023-01-05:
I, 
i've tried all the examples given but my ttgo has allways a black screen..
no errors when upload the sketches... but nothing happen
i've unpluged and re pluged the device several time...

what can i do now ?
thanks all

Shopworker, 2023-01-14:
Maybe you will have to flash the firmware new.
This video helped for me:
https://www.youtube.com/watch?v=zledBB6UPqE&ab_channel=ShotokuTech

defaliz, 2023-01-22:
thank shopworker..
i've resolved my problem.. it was a library problem..
regards

dmognaschi, 2023-01-29:
I have the same problem. Just updatd the firmware, same issue.
I'm using TFT_eSPI library.

In 'User_Setup_Select.h'
- commented #include <User_Setup.h>
- decommented '#include <User_Setups/Setup25_TTGO_T_Display.h>'

Even the example 'Factory_Test.ino' gives only a black screen. @defaliz any suggestion?

dmeylman, 2023-01-31:
@dmognaschi, your setup works for me...

closed 2023-02-12 by lewisxhe

dmognaschi, 2023-02-12:
yes @lewisxhe , it was an hw issue due to a defective device. Thank's for your time! Davide