Xinyuan-LilyGO/TTGO-T-Display #104: OTA sketch update doesn't work with the T-Display (but works fine with other boards)

Rollmops67, 2023-02-14:
Hello !

I'm using OTA sketch updates on projects with Seeed XIAO ESP32-C3 boards, and it works flawlessy.

Today I wanted to test OTA upload on the T-Display (based on the same "basicOTA.ino" sketch coming with the ESP32 pack).
But it doesn't work,
After having uploaded "basicOTA.ino" a first time through USB, when I try to upload it a second time through OTA, after compiling Arduino IDE 1.8.19 gives me these errors (I'm too noobish to understand the meaning) :

```
Sending invitation to 192.168.1.152 
Uploading: [                                                            ] 0% 
11:10:23 [ERROR]: Error Uploading
11:10:23 [ERROR]: Error Uploading

processing.app.SerialException: Erreur d'ouverture du port série « 192.168.1.152 ».
	at processing.app.Serial.<init>(Serial.java:152)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$2.<init>(SerialMonitor.java:132)
	at processing.app.SerialMonitor.open(SerialMonitor.java:132)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:132)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2126)
	at processing.app.Editor.access$1300(Editor.java:116)
	at processing.app.Editor$UploadHandler.run(Editor.java:2095)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - 192.168.1.152; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.<init>(Serial.java:141)
	... 8 more
Erreur d'ouverture du port série « 192.168.1.152 ».
```
I allready made somme projects with Wifi with the T-Display, no probleme on this side, the Wifi is connecting.
If somebody has a hint, thank you !

Roland

lewisxhe, 2023-03-13:
This has nothing to do with the board. The reason for the failure may be that the partition table you selected does not contain OTA partitions, or the network condition. Please check

Rollmops67, 2023-03-13:
Thank you for the answer !
I have little problems to understand when you say "the partition table you selected".
Can you be a little more precise ?
Thank you !
Roland

lewisxhe, 2023-03-13:
[image]
As shown in the figure, you can open debug to view the cause of failure and select the partition table correctly

Rollmops67, 2023-03-13:
OK !
It works now, many thanks to you !
With the last ESP32 boards update (2.0.7) for Arduino IDE, the only choice of memory partition is
`16 MB Flash (3MB APP/9.9 MB FATFS)` but it works.
Many thanks again !

Roland

closed 2023-03-14 by lewisxhe