Xinyuan-LilyGO/TTGO-T-Display #115: T-Display Blank Screen

pr0me7heu5, 2023-06-07:
Hello, 

I have bought and used several of these devices and in the past i have found them delightful and very straightforward to use. However, coming back after a few months to work on a new project I have found that although the serial port is behaving as normal, and the GPIO's work, the screen is **totally blank**, and even the test programs no longer work. 

Obviously i have tried various forks, (Bodmer, Xinuan) but the display is steadfastly black. I tried recompiling my favourite clock, and now it too has a black screen. There seems to be no light at all coming from it.

I'm getting worried. What have I done, and how do i get my lovely screens back, please?

Arduino 1.8.19
TFT_eSPI 2.2.20

pr0me7heu5, 2023-06-07:
I just did a clean install with arduino 2.1.0 and new libraries, and still no display. Have I broken all these displays? I tried with a brand new t-display, the test program was perfectly functioning with colours and text, but when I uploaded my simple code to it, the display was black again. Serial writes and GPIOs still function. If I had any hair i would be tearing it out....

pr0me7heu5, 2023-06-09:
Seems i was using the wrong user setup file. i used #25 and it started working again.

closed 2023-06-09 by pr0me7heu5

DawsonLeBrown, 2023-11-07:
@pr0me7heu5 - The issue you have linked to here doesn't seem to be related to your issue. I assume that you intended to link to something else, and as I too am unable to get anything to display on my screen, I'd love to know how you solved it.

uSlackr, 2024-02-23:
@DawsonLeBrown  look in [here:](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/tree/master/TFT_eSPI/User_Setups)