Xinyuan-LilyGO/TTGO-T-Display #121: How to make circuitpython work?

zwn, 2023-10-12:
I am trying [circuitpython 8.2.6](https://downloads.circuitpython.org/bin/lilygo_ttgo_tdisplay_esp32_16m/en_US/adafruit-circuitpython-lilygo_ttgo_tdisplay_esp32_16m-en_US-8.2.6.bin) firmware but it is not working for me.

I see that support for the board has been added to circuitpython here
- https://github.com/adafruit/circuitpython/pull/7750

so it seems that someone has it working.

I am able to flash the firmware (using esptool or the firmware downloader from this repository) but that's about it. The blue LED is blinking furiously, the LCD is off and the only data on the serial line is this:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12748
load:0x40080400,len:3024
entry 0x400805dc
```

So, what am I doing wrong? Is it even supposed to work?

zwn, 2023-10-12:
When I flash the [testing firmware](https://github.com/Xinyuan-LilyGO/TTGO-T-Display/tree/0ea22274252cc80bc6b572ccc201637f983c65aa/firmware), it works, so it does suggest the board and the flashing process is working.

zwn, 2023-10-13:
So the problem is that my board is 4MB variant and the board supported by circuitpython is 16MB variant. I've found out by trying different images to see which one will work ([adafruit_feather_huzzah32](https://github.com/adafruit/circuitpython/tree/main/ports/espressif/boards/adafruit_feather_huzzah32) worked). The difference was the info about SPI memory. The working board has

```
CIRCUITPY_ESP_FLASH_MODE = dio
CIRCUITPY_ESP_FLASH_FREQ = 40m
CIRCUITPY_ESP_FLASH_SIZE = 4MB
```

while the non-working has

```
CIRCUITPY_ESP_FLASH_MODE = qio
CIRCUITPY_ESP_FLASH_FREQ = 80m
CIRCUITPY_ESP_FLASH_SIZE = 16MB
```

Creating my own board in `ports/espressif/boards` with the right config and building circuitpython gave me a working firmware.

asiorgad, 2023-11-27:
@zwn Can you provide the firmware please ?