Xinyuan-LilyGO/TTGO-T-Display #122: blank screen but still uploads code

JandGprojects, 2023-10-15:
on my esp32 ttgo t-display it sees my computer and uploads code but the screen is blank when using my code that worked before and even the sketches. By the way I use the TFT_eSPI library and the cable still works since it uploads and connects to the board.
I have tried many solutions I could find on the web but nothing has so far worked, does anyone know a answer to my problem?

DawsonLeBrown, 2023-11-07:
I've got the same issue.

I have uploaded a simple test script to the board which once loaded to the board is able to output text to the serial monitor, and blink a led on GPIO 15 - but I am unable to get anything on the screen.

example script:

```c
#include "TFT_eSPI.h"

TFT_eSPI tft= TFT_eSPI();
void setup()
{
    Serial.begin(115200);
    Serial.println("Setup Starting");
    pinMode(15, OUTPUT);
    delay(100);
    digitalWrite(15, LOW);

    tft.init();

    delay(100);
    Serial.println("Setup Done");
}
int i=0; 
void loop()
{
    tft.setRotation(3);
    tft.fillScreen(TFT_ORANGE);
    i++;
    Serial.println("loop");
    digitalWrite(15, HIGH);
    delay(25);
    digitalWrite(15, LOW);
    // Wait a bit before going again.
    delay(5000);
    Serial.println("again");
}
```

aleonnet, 2023-11-08:
I have the same problem. Still couldn't run any example using display.

Av2939, 2023-12-16:
> I had the same problem. In my case I removed the old Arduino IDE and installed a new one(2.2.1). Also I'm using a more recent version of the library TFT_eSPI, check that at https://github.com/Bodmer/TFT_eSPI/releases or type TFT_eSPI on Arduino library manager.
> 
> Edit the User_Setup_Select.h file on /Users/./Documents/Arduino/libraries/TFT_eSPI
> 
> Comment the line (add // to th beginning of the line) #include <User_Setup.h> // Default setup is root library folder
> 
> And uncomment the line //#include <User_Setups/Setup25_TTGO_T_Display.h> // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT

Yup this fixed it for me.

rusosnith, 2025-04-12:
I ended up in the same corner as you but couldn make it work.
The esp32 works (prints in the serial monitor, turns on an off the backlight) but the tft wont show anything