Xinyuan-LilyGO/TTGO-T-Display #125: TTGO T-Display v1.1 serial port problems

jeffryr, 2024-03-05:
I bought a few TTGO T-Display v1.1 modules awhile back and put them in a drawer.  When I exhausted my supply of the older boards, I discovered that I couldn't communicate with the newer boards.  Debian 12 doesn't recognize the CH9102F USB to UART Bridge controller.  I built the driver from https://github.com/WCHSoftGroup/ch343ser_linux, did the usual modprobe ch343 and tried this with two different v1.1 devices.  They both boot and run the factory firmware but never show up as a serial device, dmesg reveals the following:
```
[   90.808423] ch343: loading out-of-tree module taints kernel.
[   90.808528] ch343: module verification failed: signature and/or required key missing - tainting kernel 
[   90.810623] usbcore: registered new interface driver usb_ch343
[   90.810626] ch343: USB serial driver for ch342/ch343/ch344/ch347/ch9101/ch9102/ch9103/ch9104, etc.
[   90.810628] ch343: V1.8 On 2024.02
[  118.167362] usb 2-1.1: new full-speed USB device number 4 using ehci-pci
[  118.247365] usb 2-1.1: device descriptor read/64, error -32
[  118.435388] usb 2-1.1: device descriptor read/64, error -32
[  118.623367] usb 2-1.1: new full-speed USB device number 5 using ehci-pci
[  118.703358] usb 2-1.1: device descriptor read/64, error -32
[  118.891385] usb 2-1.1: device descriptor read/64, error -32
[  118.999644] usb 2-1-port1: attempt power cycle
[  119.603361] usb 2-1.1: new full-speed USB device number 6 using ehci-pci
[  120.019370] usb 2-1.1: device not accepting address 6, error -32
[  120.099362] usb 2-1.1: new full-speed USB device number 7 using ehci-pci
[  120.515369] usb 2-1.1: device not accepting address 7, error -32
[  120.515625] usb 2-1-port1: unable to enumerate USB device
[  189.079929] usb 2-1-port1: Cannot enable. Maybe the USB cable is bad?
```
This is with a known good cable.  Also tried loading the driver on Windows 10, it just reports a malfunction when the device is plugged in.  Has anyone gotten these to work with any OS?

degagevip, 2024-05-07:
我也碰到了相同问题，角上有一个蓝色的灯在闪，我焊在了PCB上面拆下来很难。

JackJack261, 2024-06-04:
Im also having a similar issue I think. I tried uploading a sketch, and ever since my PC has trouble recognizing the device. I can't get the device to connect to any ports that Arduino IDE can recognize, even after trying to install the driver recommended in the README. Have you found a solution yet?

[image]

Image of the device in my device manager for reference