Xinyuan-LilyGO/TTGO-T-Display #127: Getting GPIO Error

IBakeCookies, 2024-04-27:
When I try to upload I get this error

```
In file included from c:\Users\****\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:48,
                 from c:\Users\****\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
c:\Users\****\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In function 'void dc_callback(spi_transaction_t*)':
c:\Users\****\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:115:22: error: 'GPIO' was not declared in this scope
  115 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |    
```

in User_Setup.h I have
`#define ILI9341_DRIVER`

in User_Setup_Select.h I have
`#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT
`

Also for some reason the FactoryTest directory does exist, but Arduino never sees it

I tried looking around and using other setups, which will make the device work, but not the screen...

Mr-PauI, 2024-07-10:
I am having this exact same issue.

bagonyi, 2024-07-16:
I have fixed this by downgrading the `esp32` board manager to version `2.0.17` (from 3.x) in the Arduino IDE:

[image]

I have also removed the `TFT_eSPI` library provided by this GitHub repo, and installed the stock version of the `TFT_eSPI` package via the Library Manager:

[image]

After installing the library, I edited `~/Documents/Arduino/libraries/TFT_eSPI/User_Setup_Select.h`:

1. Commented out line `#include <User_Setup.h>           // Default setup is root library folder`
2. Uncommented line `#include <User_Setups/Setup25_TTGO_T_Display.h>    // Setup file for ESP32 and TTGO T-Display ST7789V SPI bus TFT`

[image]

albfan, 2024-11-13:
downgrading to esp32 2.0.17 was enough for me. I need to choose ESP32 Dev Module as using lilygo T-Display there's some error with flags