Xinyuan-LilyGO/TTGO-T-Display #128: The T-Display code is broken

stevexyz, 2024-09-26:
I just bought the TTGO T-Display v1.1, recognized in Arduino as:

> BN: LilyGo T-Display
> VID: 1a86
> PID: 55d4
> SN: 589A009269

Installed arduino (on Linux Mint), copied this repository library TFT_eSPI to the Arduino directory and tried some video examples but none seems working. Below the output of the compilation:

```
WARNING: Category 'Sound' in library ESP_I2S is not valid. Setting to 'Uncategorized'
WARNING: Category 'Sensor' in library ESP_NOW is not valid. Setting to 'Uncategorized'
WARNING: Category 'Sound' in library ESP_SR is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP Insights is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library TFLite Micro is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
In file included from /home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.h:48,
                 from /home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:17:
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In function 'void dc_callback(spi_transaction_t*)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:115:22: error: 'GPIO' was not declared in this scope
  115 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:652:27: note: in expansion of macro 'DC_D'
  652 |   if ((bool)spi_tx->user) DC_D;
      |                           ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:114:22: error: 'GPIO' was not declared in this scope
  114 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:653:8: note: in expansion of macro 'DC_C'
  653 |   else DC_C;
      |        ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_write()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:159:22: error: 'GPIO' was not declared in this scope
  159 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:39:5: note: in expansion of macro 'CS_L'
   39 |     CS_L;
      |     ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_write()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:160:22: error: 'GPIO' was not declared in this scope
  160 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:56:7: note: in expansion of macro 'CS_H'
   56 |       CS_H;
      |       ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::begin_tft_read()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:159:22: error: 'GPIO' was not declared in this scope
  159 |         #define CS_L GPIO.out_w1tc = (1 << TFT_CS); GPIO.out_w1tc = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:77:5: note: in expansion of macro 'CS_L'
   77 |     CS_L;
      |     ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::end_tft_read()':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:160:22: error: 'GPIO' was not declared in this scope
  160 |         #define CS_H GPIO.out_w1ts = (1 << TFT_CS)//;GPIO.out_w1ts = (1 << TFT_CS)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:97:7: note: in expansion of macro 'CS_H'
   97 |       CS_H;
      |       ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writecommand(uint8_t)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:114:22: error: 'GPIO' was not declared in this scope
  114 |         #define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)
      |                      ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:548:3: note: in expansion of macro 'DC_C'
  548 |   DC_C;
      |   ^~~~
/home/steve/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp: In member function 'void TFT_eSPI::writedata(uint8_t)':
/home/steve/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:115:22: error: 'GPIO' was not declared in this scope
  115 |         #define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)
[... 112 lines cut ...]
Hash of data verified.
Compressed 311904 bytes to 171895...
Writing at 0x00010000... (9 %)
Writing at 0x0001a6d4... (18 %)
Writing at 0x00028c45... (27 %)
Writing at 0x0002e1d2... (36 %)
Writing at 0x00033878... (45 %)
Writing at 0x000391e0... (54 %)
Writing at 0x0003e49f... (63 %)
Writing at 0x000438c5... (72 %)
Writing at 0x00048e74... (81 %)
Writing at 0x00051a9d... (90 %)
Writing at 0x0005952a... (100 %)
Wrote 311904 bytes (171895 compressed) at 0x00010000 in 3.2 seconds (effective 773.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

BUT the only board effect is to light up the display!

Did I do something wrong? In case, can you elaborate in the homepage instructions?

Thank you

Cyrille-de-Brebisson, 2024-11-03:
Hello,

I had the same issue. I moved to the esp board package 2.0.14 and it worked again...
apparently, they seem to be an issue with v 3 and above.

Cyrille

PitStone, 2025-01-18:
No need to use the fork of TFT_eSPI in this repo. I've got it working with
- TFT_eSPI 2.5.43
- board package esp2 3.1.1

Locate the file \Arduino\libraries\TFT_eSPI\User_Setup_Select.h and do:
- put a comment in front of `#ifndef USER_SETUP_LOADED`
- uncomment the line with `#include <User_Setups/Setup25_TTGO_T_Display.h>`